iotsound/iotsound #386: Spotify Plugin cannot start, due to an argument error

wusala01, 2020-12-31:
**Describe the problem**
The `spotify` plugin is unable to start due to the following error:
```
error: Unrecognized option: 'linear-volume'
```

**To Reproduce**
I cannot assure that this error is reproducable, but I deployed the currently latest commit (31d2ab1) to balenaCloud.
balenaCloud is showing the release as 37c9432abe4d0d1305ddc8723f94fa2a however.

**Expected behavior**
The service should start, shouldn't it? :D

**Logs**
The following log is presented: 
```
 spotify  Starting Spotify plugin...
 spotify  Device name: XXXXXXXXXXX
 spotify  Volume normalization enabled.
 spotify  error: Unrecognized option: 'linear-volume'
 spotify  Usage: /usr/bin/librespot [options]
 spotify  
 spotify  Options:
 spotify      -c, --cache CACHE   Path to a directory where files will be cached.
 spotify          --disable-audio-cache 
 spotify                          Disable caching of the audio data.
 spotify      -n, --name NAME     Device name
 spotify          --device-type DEVICE_TYPE
 spotify                          Displayed device type
 spotify      -b, --bitrate BITRATE
 spotify                          Bitrate (96, 160 or 320). Defaults to 160
 spotify          --onevent PROGRAM
 spotify                          Run PROGRAM when playback is about to begin.
 spotify          --emit-sink-events 
 spotify                          Run program set by --onevent before sink is opened and
 spotify                          after it is closed.
 spotify      -v, --verbose       Enable verbose output
 spotify      -u, --username USERNAME
 spotify                          Username to sign in with
 spotify      -p, --password PASSWORD
 spotify                          Password
 spotify          --proxy PROXY   HTTP proxy to use when connecting
 spotify          --ap-port AP_PORT
 spotify                          Connect to AP with specified port. If no AP with that
 spotify                          port are present fallback AP will be used. Available
 spotify                          ports are usually 80, 443 and 4070
 spotify          --disable-discovery 
 spotify                          Disable discovery mode
 spotify          --backend BACKEND
 spotify                          Audio backend to use. Use '?' to list options
 spotify          --device DEVICE Audio device to use. Use '?' to list options if using
 spotify                          portaudio or alsa
 spotify          --mixer MIXER   Mixer to use (alsa or softvol)
 spotify      -m, --mixer-name MIXER_NAME
 spotify                          Alsa mixer name, e.g "PCM" or "Master". Defaults to
 spotify                          'PCM'
 spotify          --mixer-card MIXER_CARD
 spotify                          Alsa mixer card, e.g "hw:0" or similar from `aplay
 spotify                          -l`. Defaults to 'default'
 spotify          --mixer-index MIXER_INDEX
 spotify                          Alsa mixer index, Index of the cards mixer. Defaults
 spotify                          to 0
 spotify          --mixer-linear-volume 
 spotify                          Disable alsa's mapped volume scale (cubic). Default
 spotify                          false
 spotify          --initial-volume VOLUME
 spotify                          Initial volume in %, once connected (must be from 0 to
 spotify                          100)
 spotify          --zeroconf-port ZEROCONF_PORT
 spotify                          The port the internal server advertised over zeroconf
 spotify                          uses.
 spotify          --enable-volume-normalisation 
 spotify                          Play all tracks at the same volume
 spotify          --normalisation-pregain PREGAIN
 spotify                          Pregain (dB) applied by volume normalisation
 spotify          --volume-ctrl VOLUME_CTRL
 spotify                          Volume control type - [linear, log, fixed]. Default is
 spotify                          logarithmic
 spotify          --autoplay      autoplay similar songs when your music ends.
 spotify          --disable-gapless 
 spotify                          disable gapless playback.
 spotify  
```

**Device information**
 - Board (hardware info):Raspberry Pi 3B+ and Raspberry Pi 4B 4GB
 - Device type (as seen in balenaCloud): Raspberry Pi 3 and Raspberry Pi 4
 - balenaOS version: balenaOS v2.60.1+rev1 and balenaOS v2.65.0+rev1

**Additional context**
Ok, so from the logs there are two possible options as replacement for `--linear-volume`:
1. `--mixer-linear-volume` and
2. `--volume-ctrl linear`

The first option does not seem to be documented in the [libresport wiki](https://github.com/librespot-org/librespot/wiki/Options), so I would guess option two may be preferable. But as a new user to this project, I am not sure, what the old option was targeting.

I hope I could help.

eiddor, 2020-12-31:
@wusala01 I found where the change was made in the librespot wiki [here](https://github.com/librespot-org/librespot/wiki/Options/_compare/540f796d9e026425c02350afc7865c0f4eda8b44), but I'm still poking around to figure out when the ``--linear-volume`` flag itself was removed from the code.

I'll keep digging to see if I can find it, but either way, if you want to edit ```plugins/spotifiy/start.sh``` and replace ```--linear-volume``` with ```--volume-ctrl linear``` to test that would be a good start.

eiddor, 2020-12-31:
Interestingly enough, the change was made back in July [here](https://github.com/librespot-org/librespot/commit/f0b3b2c7e85e4d48ab940f170edd76cae87c68f9#diff-2d773e863785a6c1bf10889d0670511a28a889e57c9080caa9433e606f9a60c3) - I don't know enough about how things come together between balena-sound and librespot to figure out why it's suddenly not working.

Still digging around as a learning exercise, but I may just submit a PR to change that option in ```start.sh``` if you confirm that it's working for you.  I'm sure there's something I'm missing that would suddenly make this an issue.

shawaj, 2020-12-31:
@wusala01 @eiddor I was experiencing this same behaviour last night as well. I thought it was me messing something up whilst trying to add a new plugin for Roon, but seeing as you are having the same issues with the "vanilla" version it seems it's not just me!

Testing some changes now too...

wusala01, 2020-12-31:
Got it to work: https://github.com/wusala01/balena-sound

eiddor, 2020-12-31:
Thanks, @shawaj - I was going to attempt to test and then open a PR, but let me know if you'd rather do it.

I just tested with the latest and I'm having the same problem.  Changing to ``--volume-ctrl linear`` works fine.

It looks like raspotify caught up [here](https://github.com/dtcooper/raspotify/commit/2594e1dff9a6b710a29673717a726bca014a82d8) last night.

shawaj, 2020-12-31:
Yeah @eiddor I was going to say that I guess what had happened is that the .deb repository only just got updated with the new changes. And it looks like that's correct! Good find.

eiddor, 2020-12-31:
Thanks, @shawaj!

closed 2020-12-31 by bulldozer-balena[bot]

shawaj, 2020-12-31:
thanks to you @eiddor and @wusala01 for testing and reporting.

For reference this also relates to this forum thread - https://forums.balena.io/t/spotify-error-error-unrecognized-option-linear-volume/230187

commit 21105d46 (2020-12-31): spotify: fix linear volume control issue

rasptify updated their librepsot debs to pull in the new changes in librespot to move away from --linear-volume and to either --mixer-linear-volume or --volume-ctrl linear. The latter seems to be the preferred option judging by their latest docs update.

Change-type: patch
Connects-to: #386
Closes: #386
PR #387 (2020-12-31): spotify: fix linear volume control issue
rasptify updated their librepsot debs to pull in the new changes in librespot to move away from --linear-volume and to either --mixer-linear-volume or --volume-ctrl linear. The latter seems to be the preferred option judging by their latest docs update.

Change-type: patch
Connects-to: #386
Closes: #386