iotsound/iotsound #394: docs: plugin creation documentation

shawaj, 2021-01-01:
Having been experimenting with the plugins setup in #388 and how it works, I was thinking that we could maybe have a `docs/07-plugins.md` file or modify the plugins section of the [`03-customization.md`](https://github.com/balenalabs/balena-sound/blob/master/docs/03-customization.md#plugins) file to incorporate some information about the process / steps for building your own plugin and how everything ties together.

This might help other potential contributors, and lead to more collaborations on PRs etc.

Happy to have a first stab at it, if you think it would be useful?

eiddor, 2021-01-01:
I think this is a great idea - I don't know that I'd ever be talented enough to build any of my own, but it would _definitely_ help me learn the architecture better while troubleshooting/helping others with the issues as I've been trying to do.

shawaj, 2021-01-01:
@eiddor exactly what I was thinking! I am new to the balena team, and on the hardware/distribution side so was working on the RoonBridge plugin linked above in part because Roon is great software but also to teach myself more about balenaCloud / balenablocks / balenaSound and how everything ties together...much easier to do this with an actual project I thought 😄 

Have hit a number of road blocks, but that is not necessarily a bad thing - failing helps to learn things sometimes I guess!

But some of the things surrounding plugins and how they interact with the `audio` block could quite easily be documented I feel, and help people like you and I to get going a bit quicker. Haha.

eiddor, 2021-01-01:
I would definitely be to willing to help more I had a good starting point like this.

To be honest, maybe even take it up a level to ```docs/07-architecture.md``` to explain how all of the balena* components work together, too.  

I know I've had to do a lot of Googling/tab opening to try to isolate things like how ```librespot``` related to ```raspotify``` which related to the ```spotify``` plugin (I'm still missing something from the other day).  But I also had to do some digging to figure out an alsa fix in balenaSound that @tmigone fixed for me a few weeks ago.

shawaj, 2021-01-02:
@eiddor I just discovered we already have all this info in [this doc on arhitecture](https://github.com/balenalabs/balena-sound/blob/master/ARCHITECTURE.md)

Plugins info here - https://github.com/balenalabs/balena-sound/blob/master/ARCHITECTURE.md#plugins-1

Guess we just need to link to it more clearly somewhere.

Also this is helpful - https://github.com/balenablocks/audio#usage

Guess it could be fleshed out with some more examples etc potentially

eiddor, 2021-01-02:
Ahhh, extremely helpful - I'll have a look at them later.  I agree that making it more prominent along with some examples would help others even more.

Thanks, @shawaj !

shawaj, 2021-01-04:
Also @eiddor take a look at #388 and #399 where I have been working on a couple of new plugins. My hackery might help to show how some bits work. Although right now it is not documented very well haha

eiddor, 2021-01-04:
@shawaj Believe me, I've been seeing every commit as you push it!  It's a bit above my capabilities as someone who doesn't code, but it's fun to try to read/understand what you're trying to accomplish.  Very helpful!

shawaj, 2021-01-04:
@eiddor I'm not a very good coder to be honest.  Mainly do hardware and distribution but in my previous company (Pi Supply / JustBoom) I worked with a lot of stuff to do with audio cards and associated drivers. So know a little bit about the software side of that.

Mainly I'm just good at googling and copy pasting and seem to have a (sometimes frustrating) desire to find solutions to things that should just work. But I guess that's why we are all into Raspberry Pi, IoT and tinkering - always hunting the next challenge 😜👌