iotsound/iotsound #395: API for homeautomation system

Smassard82, 2021-01-01:
Hi,

I wonder if there is or will be some sort of an web API to integrate the balenaSound devices into an homeautomation system.

**Is your feature request related to a problem? Please describe.**
Integrating the balenaSound System into a homeautomation system.

**Describe the solution you'd like**
I would like to control the devices from our homeautomation system f.ex. turn the device on/off by detecting presence in a room, control the volume from a wallmounted touchpanel ...

**Describe alternatives you've considered**
/

**Additional context**
/

Regards

shawaj, 2021-01-01:
@Smassard82 perhaps something like our homeassistant project would be of interest to you? https://github.com/balenalabs-incubator/balena-homeassistant

What home automation system do you use?

Smassard82, 2021-01-02:
@shawaj 

Hi, Thank you very much for your reply.

So theoretical it is possible to control the balenaSound devices from an homeautomation system rigth?

I have an RPI running with [https://www.iobroker.net/#en/intro](url). 
Iobroker also has MQTT build-in.

Is there a documentation on how to control balenaSound?

Regards

shawaj, 2021-01-02:
It's definitely possible. You should be able to have a container running the home automation software and then use that to communicate with the other containers to do things based on actions in the home automation container.

We don't have an easy way to do it right now. You'd have to integrate it yourself. But I do agree that it seems like a great idea

tmigone, 2021-01-19:
Hey @Smassard82, the `sound-supervisor` container exposes some balenaSound functionality through an API on port 3000, I'm not sure what type of control you want to exert but perhaps this could be useful for your use case?

We haven't released proper documentation for this API as it's currently still "experimental" (meaning it might change in the future) but it is being used by multiple services internally by balenaSound since v3.0. It should be fairly easy however for you to reverse engineer the API endpoints as they are quite simple, code is here: https://github.com/balenalabs/balena-sound/blob/master/core/sound-supervisor/src/SoundAPI.ts

Let me know if this is useful to you. I'm happy to help you getting it to work, and we can even add more endpoints to the API to expose other stuff you might need if it's possible.

closed 2022-04-05 by tmigone