iotsound/iotsound #396: meta: build flags for plugins?

shawaj, 2021-01-01:
I was thinking that as the number of plugins for balenaSound grows, it may be that rather than just turning off a plugin using `SOUND_DISABLE_PLUGIN` options in balenaCloud, it might be better to have it so that you can add some kind of build flags / options that stop the builder from building certain plugins that you don't need.

For example, by default it could add all plugins, but if you only needed the spotify and bluetooth plugins you could add `--plugin spotify bluetooth` or something like that. Or if you want to disable some plugins you could use `--plugin-disable upnp roon` for example.

This would have the added benefit of smaller image sizes being built and pulled down to the devices as well - meaning quicker builds and downloads/updates.

I am not sure if this is possible to do with flags, or perhaps via some environment variables on the application?

AlexProgrammerDE, 2021-01-02:
@shawaj i assume you can partially already exclude containers through the dockerignore file? I also think this should be forwarded to the cli github repository and maybe renamed to --exclude-containers? Here is a link to the cli: https://github.com/balena-io/balena-cli

shawaj, 2021-01-02:
@AlexProgrammerDE yeah I agree I think this has wider use cases than just balenaSound so probably makes more sense in the CLI

Would be nice to somehow deal with this through the balenaCloud dashboard when using the deploy button as well. So you can do it all without touching the terminal at all.

tmigone, 2021-01-19:
Yes! A growing concern of mine, as we get more plugins the project gets bigger and bigger, it takes more time to build, to deploy and for low end devices it could mean a performance impact. If I had to guess most people use one or maybe two plugins so we should probably do something about it. @shawaj I'll tag this for a labs brainstorm!

shawaj, 2021-01-19:
@tmigone I have a feeling I posted this somewhere in a FD thread about the CLI, but I can't seem to find it now. Will see if I can dig it out!

tmigone, 2021-04-20:
Superseded by https://github.com/balenalabs/balena-sound/issues/412

closed 2021-04-20 by tmigone