iotsound/iotsound #397: add more dacs :-)

shawaj, 2021-01-02:
Some low hanging fruit dacs to pull in once [this PR](https://github.com/balena-os/balena-raspberrypi/pull/585) on balenaOS is merged...

- [ ] [audioinjector zero / stereo / octo / ultra / pro / isolated etc..](http://www.audioinjector.net/)
- [ ] merus audio / infineon amp #383 
- [ ] justboom-both (allows the justboom dac or amp to be used simultaneously with the digi)
- [ ] [allo boss2 / digione / katana / piano / piano plus](https://allo.com/sparky-dac.html)
- [ ] [IQaudIO codec zero](https://www.raspberrypi.org/products/iqaudio-codec-zero/)
- [ ] [hifiberry dacplusadc / adcpro /dsp / hd](https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/)
- [ ] [pifi 40](https://pifi.bz/product/pifi-amp-240/)
- [ ] [Fe-pi dac](https://www.tindie.com/products/kuupaz/fe-pi-audio-z-v2/)
- [ ] [apple-pi / pecan pi dac](https://orchardaudio.com/pecanpi)
- [ ] [audiosense pi](https://gitlab.com/kakar0t/audiosense-pi)
- [ ] [Audiophonics I-Sabre ES9028Q2M](https://www.audiophonics.fr/en/dac-and-interfaces-for-raspberry-pi/audiophonics-i-sabre-dac-es9028q2m-raspberry-pi-3-2-a-b-spdif-i2s-p-12105.html)
- [ ] [terraberry dac](https://www.terratechnos.com/products/terraberrydac/index-e.html)
- [ ] [various suptronics dacs](http://www.suptronics.com/boards.html)
- [ ] [super audio board](https://github.com/whollender/SuperAudioBoard)
- [ ] [bassfly / bassfly-mic](https://github.com/Darmur/bassfly-uhat)
- [ ] [max98357a](https://www.adafruit.com/product/3006)
- [ ] pibell - can't find link for this
- [ ] mbed-dac - can't find link for this
- [ ] [audu7002-simple](https://www.analog.com/en/products/adau7002.html#)

These should already work *without* the above patch but need testing:
- [ ] [Google AIY voiceHAT v1/v2](https://forums.balena.io/t/almost-got-google-aiy-dac-working/206819/13)
- [ ] [phat dac / phat beat / picade hat / speaker phat](https://shop.pimoroni.com/products/phat-dac) - these will already work but no longer in production so not a huge amount of point
- [ ] [dionauadio dacs / amps](https://dionaudio.nl/categorie/producten/streaming/)
- [ ] [raspiaudio dac range](https://raspiaudio.com/raspiaudio)
- [ ] [osa dacberry one / pro](https://www.osaelectronics.com/product/dacberry-pro/)
- [ ] [nanosound dac](https://nanomesher.com/nanosounddac/?gclid=CjwKCAiArbv_BRA8EiwAYGs23Eaol-qncGt1PYGGgKrnx-XO9JHNh8UGF395L2HBRnQR-2vrg4v43RoC5-0QAvD_BwE)
- [ ] [pi 2 design 502 / 502 pro / 503](https://www.pi2design.com/502dac.html)
- [ ] [mamboberry dac](https://collybia.com/product/mamboberry-hd-dac-192khz-24bit/)
- [ ] [HifiBox dac](https://m.aliexpress.com/item/32893484644.html)
- [ ] [generic rpi-dac](https://www.runeaudio.com/forum/generic-i-s-es9018k2m-dac-board-t6177.html)
- [ ] [rpi-cirrus-wm5102](http://www.farnell.com/datasheets/1928993.pdf) - discontinued so maybe no point
- [ ] [red rocks audio digidac1 - rra-digidac1-wm8741-audio]()

These need other drivers that aren't currently in the mainline / upstream raspi kernel and need further investigation:
- [ ] [bassowl](https://github.com/Darmur/bassfly-uhat)
- [ ] [tau dac](http://taudac.com/)
- [ ] [raspidac3](https://www.grulich.eu/files/RaspiDACv3.pdf)
- [ ] [WM8960 Audio HAT](https://www.waveshare.com/wiki/WM8960_Audio_HAT) - #264 
- [ ] [respeaker 2 / 4 linear / 6 circular](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/) - #166 
- [ ] [sabreberry32](https://nw-electric.way-nifty.com/blog/sb32j.html)
- [ ] [hifibunny q2m](https://www.diyaudio.com/forums/digital-line-level/314935-es9038q2m-board.html?s=03ad118356763b9d3de42435ae78f4c4)

There is quite a good list of compatible dacs for Raspberry Pi / Odroid / Tinkerboards [on the volumio docs website](https://volumio.github.io/docs/I2S_DACs/Adding_Compatibility_to_your_DAC.html). As a side note, most of the DACs that can work on Raspberry Pi can also work on the tinkerboards but they need to be requested to add support on their end (although the tinkerboards have quite good on board audio compared to the Pi).

Some other useful info for future:

- [Tinkerboard dacs](https://tinkerboarding.co.uk/wiki/index.php?title=QVL#Audio_DAC_.28GPIO_HAT.29)
- [odroid dacs](https://www.hardkernel.com/shop/hifi-shield-for-c2-c1/)
- [beaglebone capes](https://elinux.org/Beagleboard:BeagleBone_Capes)
- [list of dacs with chip type listed](https://www.symphonic-mpd.com/forum/topic/6/i2s-dac%E5%AF%BE%E5%BF%9C%E8%A1%A8-i2s-dac-driver-compatibility-table)

tmigone, 2021-01-19:
Nice, more DACS is 🚀

shawaj, 2021-01-19:
@tmigone I happen to have a good selection of these, for R&D reasons from doing JustBoom stuff. I need to do an inventory update on my side then will update this list with what I have!!

tmigone, 2021-01-19:
I thought I had a lot of DACs but yeah yours is on another level for sure :D

shawaj, 2021-01-19:
haha @tmigone yeah - I say r&d - but actually most of them are either in my house, friends and family houses, warehouse, shed etc haha. Hence my interest in balenaSound!

If you ever want any JustBoom stuff - I know someone who can ship you some on the house 😜

MatteoEng, 2021-03-18:
Hi, I have an hifiberry dac+dsp, but there is a way with balena to program the DSP? 
Thanks

tmigone, 2021-04-01:
@shawaj I noticed the linked PR is now closed, does that mean we can progress this? I know this is a huge list so don't worry about testing them. I also noticed we have a huge backlog of new DAC requests so I'm looking on ways to reduce that too. 

There is no formal process for adding DACs but this is what I've been doing so far. When someone requests support for a DAC, I figure out the dtoverlay and have the user test it. The DAC won't get added to the supported list until we get at least one user testing it, then it's fair game.

I figure we can add a new subsection with untested DACs and list there all the ones from your list that need testing with the appropriate dtoverlay. We might get some ppl to confirm they work if we put it out there. I also kindof like the json file volumio has so I'm tempted to do the same for balenaSound so it's easier to edit. Just need to figure out if setting up some templating is worth the trouble or not.

tmigone, 2021-04-20:
Superseded by https://github.com/balenalabs/balena-sound/issues/439

closed 2021-04-20 by tmigone