iotsound/iotsound #400: Raspberry Pi 4 not detecting SD Card with balenaOS

willtherussian, 2021-01-04:
I've flashed it many times and still doesn't work. I reflashed the raspberry os back on and that works so the SD card is fine.

shawaj, 2021-01-04:
@willtherussian where are you getting the image for balenaOS?

willtherussian, 2021-01-04:
I dont know if this is what you meant but from here 
https://dashboard.balena-cloud.com/deploy?repoUrl=https://github.com/balenalabs/balena-sound&d_id=176c68607a81cc-0ad06bb9f71268-c791039-1fa400-176c68607a97ff

willtherussian, 2021-01-04:
@shawaj 

> I dont know if this is what you meant but from here
> https://dashboard.balena-cloud.com/deploy?repoUrl=https://github.com/balenalabs/balena-sound&d_id=176c68607a81cc-0ad06bb9f71268-c791039-1fa400-176c68607a97ff

shawaj, 2021-01-04:
Ok and on that screen do you select the Raspberry Pi 4 as the device type?

I have it working fine on a Pi 4, so ideally just need to have the exact steps you are following to try and reproduce the issue.

Also how do you know it isn't working?

willtherussian, 2021-01-04:
Yes I select the Raspberry Pi 4 as the device type.

I can tell its not working as the ACT led doesn't light up and tried hooking it up to a monitor and there was nothing detected.

shawaj, 2021-01-05:
Have you tried downloading the image again? As it sounds like it maybe got corrupted on download or something like that.

willtherussian, 2021-01-05:
Yes, I've tried about 4 times

shawaj, 2021-01-05:
I have just tried this again on my end from scratch and it works. So there must be something you are not doing right.

Can you download the image from here and see if that boots to a splash screen? https://www.balena.io/os/

willtherussian, 2021-01-07:
I tried that and it doesn't work either

tmigone, 2021-01-19:
Hey @willtherussian, if you are still having trouble I suggest you post on our forums (https://forums.balena.io/). From experience this can be caused by many things: a problem with the SD card, the OS image you downloaded, network settings, etc. We have multiple support agents and community members on our forums that will be able to promptly assist you.

closed 2021-01-19 by tmigone