iotsound/iotsound #402: Intermittent DAC initialization (power up to power up)

jcat2, 2021-01-06:
**Describe the problem**
From power to power up the output of the device changes.   On one power up it will be the headphone jack.   On the next power up it will be DAC output.   No settings are changed between power ups or application restarts.   It appears that the DAC is not always being detected.   Most of the time the DAC is not being detected or configured as an output.

**To Reproduce**
Steps to reproduce the behavior:
1. Restart the application or reboot the Pi with the HiFiBerry DAC2 HD and appropriate settings

**Expected behavior**
Expectation is that the DAC would always be detected and the preferred output

**Screenshots**
Below are the log traces 
*Run with DAC not being detected*
```
05.01.21 23:04:06 (-0500)  audio  --- Audio ---
05.01.21 23:04:06 (-0500)  audio  Starting audio service with settings:
05.01.21 23:04:06 (-0500)  audio  - Pulse log level: NOTICE
05.01.21 23:04:06 (-0500)  audio  - Default output: AUTO
05.01.21 23:04:06 (-0500)  audio  
05.01.21 23:04:06 (-0500)  audio  Detected audio cards:
05.01.21 23:04:06 (-0500)  audio  0 bcm2835-jack bcm2835_headphonbcm2835Headphones-bcm2835Headphones
05.01.21 23:04:06 (-0500)  audio  
05.01.21 23:04:06 (-0500)  audio  
05.01.21 23:04:16 (-0500)  audio  Setting audio routing rules. Note that this can be changed after startup.
05.01.21 23:04:16 (-0500)  audio  Routing 'balena-sound.input' to 'snapcast'.
05.01.21 23:04:16 (-0500)  audio  Routing 'balena-sound.output' to 'alsa_output.bcm2835-jack.stereo-fallback'.
05.01.21 23:04:16 (-0500)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
05.01.21 23:04:16 (-0500)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
05.01.21 23:04:16 (-0500)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```
*Run with DAC being detected*
Log file lost and will add when recreated

**Device information**
 - Board (hardware info): Raspberry Pi 4 with HiFiBerry DAC2 HD
 - Device type (as seen in balenaCloud): Raspberry Pi 4
 - balenaOS version: balenaOS 2.65.0+rev1

**Additional context**
*Device Configuration Variables*
BALENA_HOST_CONFIG_dtoverlay hifiberry-dacplushd
*Balena Sound Version*
3.4.5

shawaj, 2021-01-06:
Hey @jcat2 when did you install the balenaSound instance? As we recently did an update in the audio block to bring in the new driver.

jcat2, 2021-01-06:
Hi @shawaj.   I did the installation last night and this morning I did a clean Balena application using the latest github repo and still had the same problem.   The additional information that I can provide is that the DAC is detected the first time the installation boots and has its configuration variable set.   Every other boot after that the DAC is not detected.   

It looks like this is isolated to balenaSound.   I've tried rebooting the same device and DAC with HiFiBerry OS and have had no issues.

shawaj, 2021-01-06:
Unfortunately I don't have one of these DACs available to test. Perhaps the people in #357 or #377 @arjan5340 @richardpringle might be able to comment on whether they are seeing the same issue at all?

Also @jcat2 have you got any device variables set? Such as the sound output one?

jcat2, 2021-01-06:
@shawaj I've tried setting the auto output to AUTO and DAC.   When DAC is set and error message stating that no DAC was found gets displayed in the logs.

shawaj, 2021-01-06:
What pi version are you using? And what power supply?

It sounds like it might be an issue with it not receiving enough power or something but it's hard to say without having one to test with so I'm just guessing at this point.

jcat2, 2021-01-07:
@shawaj I'm using a Raspberry Pi 4.   The power supply is the official Raspberry Pi 4 power supply.   

At it's face I would agree that it sounds like a power problem, however, given that it's been working reliably with HiFiBerry OS power cycle after power cycle it makes me think it's something else.

richardpringle, 2021-01-07:
@jcat2, I've seen similar issues with one of my two pi-4s with the DAC2 HD. I didn't try loading the hifiberry OS though. I'm not actually too familiar with Linux and how it scans for audio devices, you should be able to check if the card is picked up somewhere in the filesystem right (there should be a file-descriptor somewhere)? Any ideas @shawaj? This seems like a concurrency issue or something to me, as though it's possible that the OS scans for cards before the card receives power. I'm not sure if this makes sense at all though.

Is there a process that actually scans for an audio card? Or is it more like a plug-n-play USB type thing? 

I don't think it's actually a pulse-audio issue, it must be a problem with the OS?

What do you get when you `ls -la /proc/asound`?

shawaj, 2021-01-07:
@richardpringle the cards are at `/proc/asound/cards` I believe

https://alsa.opensrc.org/Proc_asound_documentation

jcat2, 2021-01-07:
@richardpringle and @shawaj I just powered up my balena-sound image and the DAC was not detected.   There is nothing listed in /proc/asound/cards.   The results of ls -la /proc/asound are found below.

```
dr-xr-xr-x   4 root root 0 Jan  7 14:13 .
dr-xr-xr-x 173 root root 0 Jan  7 14:13 ..
lrwxrwxrwx   1 root root 5 Jan  7 14:17 bcm2835-jack -> card0
dr-xr-xr-x   3 root root 0 Jan  7 14:14 card0
-r--r--r--   1 root root 0 Jan  7 14:13 cards
-r--r--r--   1 root root 0 Jan  7 14:17 devices
-r--r--r--   1 root root 0 Jan  7 14:17 modules
-r--r--r--   1 root root 0 Jan  7 14:17 pcm
dr-xr-xr-x   2 root root 0 Jan  7 14:17 seq
-r--r--r--   1 root root 0 Jan  7 14:17 timers
-r--r--r--   1 root root 0 Jan  7 14:17 version
```

On the next power up the DAC was detected.   Here are the results

```
dr-xr-xr-x   5 root root 0 Jan  7 14:22 .
dr-xr-xr-x 164 root root 0 Jan  7 14:22 ..
lrwxrwxrwx   1 root root 5 Jan  7 14:24 bcm2835-jack -> card0
dr-xr-xr-x   3 root root 0 Jan  7 14:22 card0
dr-xr-xr-x   3 root root 0 Jan  7 14:23 card1
-r--r--r--   1 root root 0 Jan  7 14:22 cards
lrwxrwxrwx   1 root root 5 Jan  7 14:24 dac -> card1
-r--r--r--   1 root root 0 Jan  7 14:24 devices
-r--r--r--   1 root root 0 Jan  7 14:24 modules
-r--r--r--   1 root root 0 Jan  7 14:24 pcm
dr-xr-xr-x   2 root root 0 Jan  7 14:24 seq
-r--r--r--   1 root root 0 Jan  7 14:24 timers
-r--r--r--   1 root root 0 Jan  7 14:24 version
```

It looks like for whatever reason the OS is not picking DAC on certain power ups.   Almost all the power ups have resulted in the DAC not being detected.

shawaj, 2021-01-08:
On the boot ups where it doesn't show, does anything show in i2cdetect -y 1 ?

It's almost like it's booting up too fast for the DAC to power up or something.

Also what happens on the times when it doesn't show if you reboot it without removing the power?

Lastly, in the hifiberry os you mentioned, does it have a /boot/config.txt file?

jcat2, 2021-01-08:
@shawaj Just had a bot up were the DAC does not show up.   Here is the log

```
08.01.21 15:10:53 (-0500)  audio  
08.01.21 15:10:53 (-0500)  audio  --- Audio ---
08.01.21 15:10:53 (-0500)  audio  Starting audio service with settings:
08.01.21 15:10:53 (-0500)  audio  - Pulse log level: NOTICE
08.01.21 15:10:53 (-0500)  audio  - Default output: AUTO
08.01.21 15:10:53 (-0500)  audio  
08.01.21 15:10:53 (-0500)  audio  Detected audio cards:
08.01.21 15:10:53 (-0500)  audio  0 Headphones bcm2835_headphonbcm2835Headphones-bcm2835Headphones
08.01.21 15:10:53 (-0500)  audio  
08.01.21 15:10:53 (-0500)  audio  
08.01.21 15:11:04 (-0500)  audio  Setting audio routing rules. Note that this can be changed after startup.
08.01.21 15:11:04 (-0500)  audio  Routing 'balena-sound.input' to 'snapcast'.
08.01.21 15:11:04 (-0500)  audio  Routing 'balena-sound.output' to 'alsa_output.bcm2835-jack.stereo-fallback'.
08.01.21 15:11:04 (-0500)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
08.01.21 15:11:05 (-0500)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
08.01.21 15:11:05 (-0500)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```

And here is the return of i2cdetect -y 1

```
root@ea08a4d325a5:/usr/src# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

I'll reinstall HiFiBerry OS to get the /boot/config/txt

jcat2, 2021-01-08:
@shawaj 

Here is the /boot/config.txt from HiFiBerry OS

```
# more config.txt
# Please note that this is only a sample, we recommend you to change it to fit
# your needs.
# You should override this file using a post-build script.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

# We always use the same names, the real used variant is selected by
# BR2_PACKAGE_RPI_FIRMWARE_{DEFAULT,X,CD} choice
start_file=start.elf
fixup_file=fixup.dat

kernel=zImage

# To use an external initramfs file
#initramfs rootfs.cpio.gz

# Disable overscan assuming the display supports displaying the full resolution
# If the text shown on the screen disappears off the edge, comment this out
disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=100
gpu_mem_512=100
gpu_mem_1024=100
# Enable I2C and SPI
dtparam=i2c=on
dtparam=spi=on
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1
dtoverlay=vc4-fkms-v3d,audio=off
dtoverlay=hifiberry-dacplushd
force_eeprom_read=0
```

shawaj, 2021-01-08:
Interesting that they are running a custom kernel. Doubt that's the issue but still.

It looks like they're also using the software i2c controller, forcing the vc4 audio off and forcing the eeprom not to be read.
```
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1
dtoverlay=vc4-fkms-v3d,audio=off
dtoverlay=hifiberry-dacplushd
force_eeprom_read=0
```

Especially the last line implies that they have got some issue with the eeprom contents or something that is incorrect.

I guess we should try replicating this in balenaCloud and see if that fixes things.

What do you have in the dtoverlay section of balenaCloud device currently?

Can you also try the following commands on the host OS shell:
```
cat /etc/asound.conf
uname -r 
uname -a
```

Also do you have any other devices attached to the Pi or is it just the DAC?

FYI pinout info here https://www.hifiberry.com/docs/hardware/gpio-usage-of-hifiberry-boards/

shawaj, 2021-01-08:
If you could do the commands above on HiFiBerry OS as well that would be amazing!

jcat2, 2021-01-08:
@shawaj 
I currently have hifiberry-dacplushd set to BALENA_HOST_CONFIG_dtoverlay.   I can add the others to the BALENA_HOST_CONFIG_dtoverlay.

Regarding the command do you want them from HiFiBerry OS or balena?   In balena there is no /etc/asound.conf file on the host shell.   uname -r reports 5.4.58 and uname -a reports Linux ce5ba44 5.4.58 #1 SMP PREEMPT Wed Dec 16 14:42:39 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux

I'll try them on HiFiBerry also.

shawaj, 2021-01-08:
Sorry the /etc/asound.conf would be on the audio container, my bad.

And yes if you could try on HiFiBerry OS that would be great.

I'm thinking this is some issue with the 5.4 kernel and the old overlay perhaps as there was some changes in the 5.4 kernel that caused issues with some overlays.

jcat2, 2021-01-09:
@shawaj 
I have no other devices connected to the pi.

jcat2, 2021-01-09:
@shawaj 
/etc/asound.conf does not exist in the sound container either.

jcat2, 2021-01-09:
Adding the HiFiBerry OS config.txt overlays and eeprom commands did not fix the issue.

jcat2, 2021-01-09:
@shawaj 

Here is the data requested on HiFiBerry OS

```
# cat /etc/asound.conf
# 
# asound.conf for exclusive audio mode
#
defaults.pcm.rate_converter "samplerate"

defaults.pcm.card 0
defaults.ctl.card 0

pcm.hifiberry {
  type hw
  card 0
  device 0
}
pcm.ttable {
 type route
 #ttable_config:stereo 0
 ttable.0.0 1
 ttable.1.1 1
 ttable.0.1 0
 ttable.1.0 0
 slave.pcm "hifiberry"
}
pcm.softvol {
  type softvol
  slave.pcm "ttable"
  control {
    name "Softvol"
    card 0
  }
  min_dB -90.2
  max_dB 0.0
}
ctl.equal {
  type equal;
}
pcm.equal {
  type equal;
  slave.pcm "plug:softvol";
}
pcm.!default { type plug; slave.pcm "softvol" } # SOFTVOL
```
```
# uname -r
5.4.77-v7l
# uname -a
Linux hifiberry 5.4.77-v7l #1 SMP Sun Dec 13 14:20:21 UTC 2020 armv7l GNU/Linux
```

richardpringle, 2021-02-27:
Any updates on this issue?

jcat2, 2021-02-28:
> Any updates on this issue?

I'm still having the same issue.   I started to move away from balena sound and just use the HiFiBerry OS seeing it works on every power up and have every similar capabilities.   The HiFiBerry maker is willing to send boards to developers if anyone from the development team wants to troubleshoot this further.