iotsound/iotsound #403: Wifi setup fails

rtcpersonal2020, 2021-01-09:
**Describe the problem**
Wifi setup fails.  RPI3 will not connect to simple wifi using WAP credentials after downloading an booting the "custom" image. 

Lots of wasted time and effort.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
 - Board (hardware info): [e.g. Raspberry Pi 3B+]
 - Device type (as seen in balenaCloud): [e.g. Raspberry Pi 3 (using 64bit OS)]
 - balenaOS version: [e.g. balenaOS v2.47.0+rev1]

**Additional context**
Add any other context about the problem here.

tmigone, 2021-01-19:
Hey @rtcpersonal2020, thanks for trying out balenaSound. If you are still having trouble with your board connecting to WiFi I suggest you post on our forums (https://forums.balena.io/), this is very likely not related to balenaSound. We are very active on the forums with multiple support agents that will help you out if needed.

closed 2021-01-19 by tmigone

rtcpersonal2020, 2021-01-19:
The volumio distro worked first time, no modifications.  It's your image.

-------
On Tue, 19 Jan 2021 06:46:12 -0800, Tomás Migone  
<notifications@github.com> wrote:

> Hey @rtcpersonal2020, thanks for trying out balenaSound. If you are
> still having trouble with your board connecting to WiFi I suggest you
> post on our forums (https://forums.balena.io/), this is [...]

-- 
Using Opera's mail client: http://www.opera.com/mail/