iotsound/iotsound #404: Multiroom, Standalone, Virtual rooms

romd87, 2021-01-12:
**Is your feature request related to a problem? Please describe.**

- I would like to have a lot of speakers connected to BalenaSound (In Living room, Kitchen, Bedroom, Bathroom, Garden)

- When I would like to stream for example Spotify music, I will choose from Spotify Device list “Living room”

- But sometimes, I would like to stream Spotify music to all speakers in whole house

So, it could be good to have some “Virtual rooms”, which will be created from existing Standalone speakers. For example I will have virtual room with name “Dayroom”, where will be connected Living room and Kitchen. Or I will have virtual room with name “Whole house”, where will be connected all speakers from house.

So when I will open device list (Spotify, Airplay, ... ), I can choose: Living room, Kitchen, Bedroom, Bathroom, Garden, Dayroom, Whole house.

**Describe the solution you'd like**
Option to create "Virtual rooms", which will be connected with existing speakers.

xvisor16, 2021-01-18:
Ow yesss this would be really amazing, but hopefully not only for spotify, but for all inputs! Although I only use spotify, some kind of consistency in the thing work would be better if you ask me! Maybe by adding some application variables like "multiroom_name_one = Downstairs" and "multiroom_name_one_clients = devicename_1, devicename_3" like style? Or maybe hook them in a way by mac address so renames aren't a nasty thing?

JurakZdenek, 2021-02-08:
Currently, when I set up the device as Master and others as Client (garden, kitchen), I see in Spotify: Living room, Garden, Kitchen. If I play music in the Living room, it will automatically start playing on all three devices. This will lose the opportunity to play music only in the Living room. It would be good to make groups as in the attached picture. That is, I would spotify to see the device: Group 1,2,3 Device 1,2,3,4 and I could choose where to play the music. Even at the cost of having to sacrifice one device as a Controller. Then it would work like Sonos.
[image]

xvisor16, 2021-02-10:
I've been thinking about this still... I understand the problem with the master-slave setup and the flexibility, the strength of Balena-Sound is there is no specific controller! The one receiving the data elects himself as a master and the others become a slave and listen/play (leaving the master/slave environment variables out of play here). My suggestion would be to keep this as is, not specifically defining one master(controller) since it would defeat the purpose/simplicity of B-S. But if we could, lets say, let the sound-supervisor choose a random master which also serves the groups/multirooms anyone wants to have and if that one becomes unavailable switch to another device as master. You could just put the group/multiroom setups in the fleet variables and who becomes master will serve those setups. The only ugly duckling will be bluetooth.... but heck that is already the ugly duckling in the multiroom setup...

But what I would would also like is the option to override locally. So lets say someone downstairs starts a multiroom stream to the livingroom, kitchen, garden and bathroom and has this playing. Now my son go's to bath and want to listen some totally different music there he has to be able to override by playing to the bathroom device... meaning all other will still play the downstairs selected audio except for bathroom which will now play different audio until stopped for more than 10 seconds or so and then join the original stream started from downstairs.

bod955, 2021-03-13:
Deciding in which rooms music should be played synchronously is for me the most important function of a multiroom setup.

There are a few alternatives to balenaSound that I have tested. Some have this feature, in that the individual players can be grouped together, but this only works for Spotify Connect and AirPlay, meaning you select a group as the output. This is perfectly adequate and workable, at least for me, since I only use Spotify or streams via AirPlay. Unfortunately, the solutions can only be used properly with an additional app and also have basic or additional problems due to the app, which greatly reduce the operation and quality. balenaSound, on the other hand, works cleanly for me via Spotify Connect and AirPlay. I therefore use Spotify, YouTube directly and don't bother with output via another app --> this is another important advantage of balenaSound.

My wishes for balenaSound are also the configuration of the groups. I see following use cases similar to previous speakers:

**Example: group with 3 devices a/b/c. 2 users A/B access devices of a group**
User A connects to the group and automatically activates multiroom operation of the connected devices. User B connects to device a from the group, which removes device a from the group and switches to standalone operation. As soon as user B disconnects from device a, the device returns to multiroom operation.

**This results in the following requirements:**
- Muliroom/standalone operation should be variable. There could be an additional operation mode "Hybrid". The device must automatically recognize which operation mode it must adopt. This allows for fixed as well as flexible multiroom configurations.
- The master within a group should be definable fixed or variable. When user B connects to the fixed master, the group switches from user A to user B. With the variable master, it is always the best performing device in the group. If the best performing device is configured as a hybrid and removed from the group by another user, the master in the group must change. (The switch will probably involve a pause in the stream, but in private this is acceptable. Also, you could avoid this with the master = fixed).

For this function I would not only donate but also pay money regularly. It would be great to collect the requirements collectively to support.

jrcarlsson, 2021-05-22:
I to would love this funktion! its also another issue opened for the same thing: https://github.com/balenalabs/balena-sound/issues/188

kculmback, 2021-06-01:
Would love to have this feature. Right now my house audio system has a single source, so whenever we want to play something through the house speakers it plays throughout the whole house.

It would be great to be able to play sound to a single room, a zone (such as 'downstairs'), or the whole house. Ideally you could also have User A play sound to a room (such as 'living room') and User B play sound to another room such as 'kitchen'). I would definitely pay a subscription for a feature such as this.

s8fabord, 2021-12-22:
I'm currently setting up a Balena multiroom project at home. Just have the 3 first Raspberry 4 with Hifiberry running. All works fine so far, but I'm astonished that there doesn't seem to be any possibility yet to select the device / or group of devices you want to stream your music...
I have 8 raspberries for 8 zones in the house. Streaming to all devices is simply not an option, I'm not alone in the house (i.e.Kids are asleep), but I want to stream to Kitchen / dining room / living room together when we have guests...
This seems to bother a few users. Is there no solution yet?

bod955, 2021-12-23:
Hey **s8fabord**, 

I have the following workaround up and running. 
- 4 zones: three Raspberry Pi 4 (kitchen, living room, office) operating in multi-room client mode ("SOUND_MODE" as "MULTI_ROOM_CLIENT")  and one Raspberry Pi 4 (bedroom) in standalone mode ("SOUND_MODE" as "STANDALONE")
- 1 additional raspberry Pi 3 (living room) operating in multi-room ("SOUND_MODE" as "MULTI_ROOM") --> this is the master with device name "All Rooms", but it has no speaker connected to it, since I do not need another one in the same room. 

This results in 5 devices. 4 of them I can operate as single output with different inputs, but when I start streaming at "All Rooms" it plays in the kitchen, living room and office. However, in the bedroom someone else could listen to different music at the same time. If you would add speakers to "All Rooms", it is not possible to only listen to music on that speakers unless the other devices with multi-room client mode are switched off.

I'm convinced, that you also could operate multiple multi-room zones with this approach like this:
- multi room Zone A (Fleet A: master a1, client b1, client c1 ...)
- multi room Zone B (Fleet B: master a2, client b2, client c2 ...)

You would have to configure as much fleets as much multi room zones you like to have. In each fleet you need a master, but you are not able to operate fleet A with fleet B synchronously. Depending on your set up, it could result in higher costs if you are adding additional PIs as Masters without speakers like I did. However, this is a practical and simple approach.

I hope that helps! Enjoy it!

s8fabord, 2021-12-24:
I had thought about something like this but hadn‘t followed the idea as I really didn’t expect to not working natively…

thanks for sharing ! 👍

beber666, 2022-01-24:
Definitely needed to choose in wich rooms I want to stream, and Balena sound will be simply perfect !

Hoggapogga, 2024-02-24:
I might have missed something but it sound like if you force set the ip of the master devices server and somehow force set the client to connect to that ip you can group them together and create "virtual rooms". 
To make this work you only have to figure out how to force connect the clients. 

If you then add the ability for all devices to have a whitelist of servers it connects to and acts like a client then you can make multiple virtual rooms / groups of speakers.