iotsound/iotsound #413: Add a plex plugin

phil-d-wilson, 2021-02-07:
For people who don't use online services for music, and use a local plex server.

allordacia, 2021-06-23:
Would also love to see an integration that worked with a local network storage option. I have been trying to get Volumio to work with Balena but it does not seem like it will end up being a viable option.

Syco54645, 2022-02-05:
This would be quite nice and would allow for something akin to airplay for Android (Plex) users.

fwicht, 2022-02-19:
It seems that remote control has been (re)introduced into [Plexamp 4.0](https://forums.plex.tv/t/plexamp-release-notes/221280/38). That means that Plexamp can now act as a `Plex audio receiver`. This could be an interesting lead as to how to implement this feature.