iotsound/iotsound #416: Synology DS Audio support

evertonpr, 2021-02-16:
Hi guys, 

I just setup my brand new Synology NAS and it brought life to my old and extensive MP3 collection.

I found the DS Audio app super nice, since I can cast to my Chromecast Audio in my living room, but what a pity Balena Sound is not listed in the devices to cast, so I cannot use it in my office! :(

It would be great if you could add the support for DS Audio. I would be happy to support with testing!

Thank you!

Everton

tmigone, 2022-04-05:
Hey, this would require balenaSound to act as a standalone Chromecast audio receiver/player. This is not something I think falls within the scope of this project as balenaSound expects to control the audio sources as well. 

What you would need is something similar to https://github.com/rahul-thakoor/balena-rpiplay but that uses Chromecast instead of apple airplay. That's not unfortunately something I've seen around so not sure where you could start.

tmigone, 2022-04-05:
By the way, this issue https://github.com/balenalabs/balena-sound/issues/102 tracks the feature request to add chromecast as an output option to balenaSound. I don't know if it's possible or when we'll look into it, but I imagine if we were to find a solution for that it would probably be very close to what you need, so maybe you can keep an eye on that other issue.

closed 2022-04-05 by tmigone