iotsound/iotsound #421: Spotify Hifi quality coming

Szubszidiaritas, 2021-02-26:
Spotify announced that it will soon introduce lossless, hifi quality streaming. Current maximum is 320 kbps, which is already supported by Balena Sound. It would be great if the new lossless feature could be used as well with Balena Sound. Is there an already existing way to ensure it, or does it require some development?

tmigone, 2021-04-20:
Hi @Szubszidiaritas, do you have a link to share about that announcement? According to docs 320kbps is high quality already, is hifi a new higher bitrate?

```
# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
```

In any case, if this is treated as a new bitrate then support for it should be automatically added once the upstream [package](https://github.com/dtcooper/raspotify) we use for spotify adds support for it. It should be done in the same way we currently support 96, 160 and 320 kbps, by setting `SOUND_SPOTIFY_BITRATE` to the new bitrate.

Szubszidiaritas, 2021-04-26:
Hi @tmigone,

Thank you for your reply! Here is one source with some details on changes in bitrate:
https://www.whathifi.com/features/spotify-hifi-is-missing-something-but-will-it-matter

The bottom line is, Spotify HiFi will enable 1411 kbps streaming in contrast with the current highest quality option, 320 kbps.
Hopefully, the solution you assume will do the trick (setting bitrate to 1411), but I am not at all knowledgeable about the technical details, ie. whether or not it will be treated as a new bitrate, or they will do it some other way. I guess we'll see how they plan to implement it when it is already launched. :)

haug-den-lucas, 2025-11-05:
> Hi, so a couple years later, lossless did indeed arrive to Spotify. It seems high time to revive this thread. How should one adjust the Spotify bitrate so as to get lossless quality? I tried 1411 instead of the current highest (320) but it did not work out.

iotsounds uses librespot in the background.
So I think we need to wait until librespot implements lossless support, see here: https://github.com/librespot-org/librespot/issues/1583

Szubszidiaritas, 2025-11-05:
> > Hi, so a couple years later, lossless did indeed arrive to Spotify. It seems high time to revive this thread. How should one adjust the Spotify bitrate so as to get lossless quality? I tried 1411 instead of the current highest (320) but it did not work out.
> 
> iotsounds uses librespot in the background. So I think we need to wait until librespot implements lossless support, see here: [librespot-org/librespot#1583](https://github.com/librespot-org/librespot/issues/1583)

Thanks for the info! Do you think if they sort it out, us iotsound users will be notified about the new feature and settings? Or perhaps someone would drop a reply here?

JaragonCR, 2026-03-02:
@AlexProgrammerDE @alanb128  this should be closed as technically unfeasible. Please see last comment here.

https://github.com/devgianlu/go-librespot/issues/204

JaragonCR, 2026-11-14:
@haug-den-lucas spotify released soloist a few moths ago, I wired it into my fork and it is now playing lossless on the pi.

haug-den-lucas, 2026-11-14:
> [@haug-den-lucas](https://github.com/haug-den-lucas) spotify released soloist a few moths ago, I wired it into my fork and it is now playing lossless on the pi.

Nice, that's amazing!
I'm actually already using a slimmed version of your fork on my device anyways, I'll check it out later.
Kind of sad that iotsound seem to be really dead, it could really benefit from your contributions...

JaragonCR, 2026-11-14:
Agreed. The market still doesn’t offer a product for what I use it for. It would have been great to have a team of all the great people that offered. :(