iotsound/iotsound #422: Pulseaudio Network Sink

heyarne, 2021-03-01:
I am on Arch Linux and I'm using Pulseaudio 14.2. Balena sound is running on a Rapberry Pi 3b+ that's equipped with a Hifiberry Amp+. I tried setting up Airplay on my Arch Linux laptop but was unsuccessful. When reading around I saw that Pulseaudio has the option to be used as a network-wide sink.

I think exposing the Pi as a Pulseaudio as a network sink could be benefitial because it uses free protocols native to both my laptop and balena sound, since from what I understand it heavily uses Pulseaudio already. This would allow me to keep the stack simpler and, by using only wifi-based protocols, also potentially avoid bluetooth issues present on the Pi 3.

borky, 2021-03-02:
i think what you want is what you will get by using the below command on your arch laptop if you have pulseaudio installed(replace the italic names with your information):
**pacmd load-module module-tunnel-sink-new server=[_yourBalenaHostnameOrIP_]:4317 sink=balena-sound.input format=s16le channels=2 rate=44100 sink_properties=device.icon_name=computer sink_name=_YourDesiredNameForTheSinkItWillCreate_ channel_map=front-left,front-right**

The above command will create a new audio tunnel (device) which you can select afterwards. You could call this command whenever you want it or when your laptop starts so that it will be always available.

heyarne, 2021-03-02:
That works great! Thanks. Maybe this could be added to the docs? It's an unmentioned method of playing audio that seems to come for free and for me it's a big plus over for example bluetooth, which is flaky at best on my laptop.