iotsound/iotsound #423: multiroom: no sound

tmigone, 2021-03-03:
**Describe the problem**
When using multiroom mode there is no sound coming out of any audio interface.
This is after deploying the latest version of balenaSound.

Szubszidiaritas, 2021-03-07:
Is it that obvious to everyone how to look at an 'attached support thread in jellyfish'? I can neither log in nor register there, and having the same, quite serious issues with my multiroom. Devices keep dropping off network, working as standalone instead of multiroom. Many times, at the start of a track, a small glitch can be heard, etc. I would be gld to access those jellyfish threads, but cannot find a way.

Szubszidiaritas, 2021-03-07:
Someteimes devices do not work at all, leaving the rest of the devices working, but some of them do not. Services stop seemingly at random, or the entire os just turns off without me initiating it. Only one of my 3 devices seem to work without issues. Now I started to rewrite the sd cards, but I would have preferred looking at those attached threads before doing so.

tmigone, 2021-03-08:
Closed by #424 and [#61](https://github.com/balenablocks/audio/pull/61).
The root cause was:
- PulseAudio version not pinned in the audio block. This was an oversight and is now fixed.
- Snapcast version not pinned in balenaSound. It was tracking the alpine edge repo to get the latest package with alsa source stream support but it was updated to a non-working version. It's now pinned but running a debian based image which increased the container size for `multiroom-server` and `multiroom-client` 3x.

closed 2021-03-08 by tmigone

tmigone, 2021-03-08:
Hi @Szubszidiaritas,

> Is it that obvious to everyone how to look at an 'attached support thread in jellyfish'? I can neither log in nor register there, and having the same, quite serious issues with my multiroom. Devices keep dropping off network, working as standalone instead of multiroom. Someteimes devices do not work at all, leaving the rest of the devices working, but some of them do not. Services stop seemingly at random, or the entire os just turns off without me initiating it. Only one of my 3 devices seem to work without issues. Now I started to reflash the OS on the sd cards, but I would have preferred looking at those attached threads before doing so. Also, many times, at the start of a track, a small glitch can be heard. Sorry for writing all these here, but it is getting rather frustrating, to be honest.

JellyFish is balena's internal support tool, it's not publicly available. Forum threads and GitHub issues are cross linked into JellyFish so it's a way for us to keep track of everything in one place. Regarding balenaSound issues, 99% of the times if you see a JellyFish ticket attached to a GitHub issue it's because there is a forum thread about it, so checking https://forums.balena.io/ is your best bet.

About your issues, the "no sound" problem should be fixed with the latest release. If you have additional problems I suggest you post on the forums too, we've got multiple support agents (myself included) on rotation helping users out.