iotsound/iotsound #430: Sources Interlock

MatteoEng, 2021-03-18:
**Is your feature request related to a problem? Please describe.**
I noticed this : if I use different sources at the same time, no one has priority, obviously I did it because  I am testing it.

**Describe the solution you'd like**
A service that Interlocks sources (bluetooth airplay spotify...)

**Describe alternatives you've considered**
It's not super important because usually a person won't use 3 sources at the same time for the same player, but would be awesome to stop a source or have some priority

**Additional context**
Add any other context or screenshots about the feature request here.

tmigone, 2021-04-20:
Hi @MatteoEng, this behaviour you describe is by design. balenaSound versions < 3.0 used to only allow a single source or audio at a time, this was modified for 3.0 onwards. At first sight I don't think we want to modify this unless there is a compelling reason to do so. Can you share a bit more what the use case would be? 

If needed this could be done by interfacing programatically with the audio block, it shouldn't be too difficult but I'm not sure how proficient you are with typescript.

tmigone, 2021-05-27:
Closing due to inactivity. Feel free to reopen.

closed 2021-05-27 by tmigone