iotsound/iotsound #432: Audio fails to start on Raspberry pi zero W + Pimoroni Pirate Audio line out DAC

drew-kun, 2021-03-24:
I am using Raspberry Pi Zero W + [Pimoroni Pirate Audio line out DAC hat](https://shop.pimoroni.com/products/pirate-audio-line-out)

BalenaOS version: balenaOS 2.54.2+rev1
Supervisor version: 11.12.4
BalenaSound version: v3.5.2

App Env Vars:
```
AUDIO_OUTPUT=DAC
BLUETOOTH_DEVICE_NAME=my-hifi
SOUND_DEVICE_NAME=my-hifi
SOUND_DISABLE_AIRPLAY= remove_this_var_to_enable_AIRPLAY
SOUND_DISABLE_SPOTIFY= remove_this_var_to_enable_SPOTIFY
SOUND_MODE= STANDALONE
SOUND_VOLUME=50
```

Device config vars:
```
BALENA_HOST_CONFIG_dtoverlay=hifiberry-dac
```

DT parameters:
[image]

All the containers are up apart of disabled multiroom-client, multiroom-server, airplay and spotify.
[image]

Full log after reboot:
```
17.03.21 21:48:13 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:18 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:23 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:29 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:34 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:35 (+0200) Killing service 'audio sha256:46b8be765bfc0fd29d814699a3d2be55a708eb7f897445db4290e06700f36c82'
17.03.21 21:48:39 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:44 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:48:49 (+0200) <audio> Waiting for sound supervisor to start
17.03.21 21:49:00 (+0200) Killed service 'audio sha256:46b8be765bfc0fd29d814699a3d2be55a708eb7f897445db4290e06700f36c82'
17.03.21 21:49:00 (+0200) Service exited 'audio sha256:46b8be765bfc0fd29d814699a3d2be55a708eb7f897445db4290e06700f36c82'
17.03.21 21:49:01 (+0200) Killing service 'bluetooth sha256:a9d5faaccb7b5d33364cdf1adbeb55a07ce0ceee0cc443a13f3fe08a5f089173'
17.03.21 21:49:25 (+0200) Killed service 'bluetooth sha256:a9d5faaccb7b5d33364cdf1adbeb55a07ce0ceee0cc443a13f3fe08a5f089173'
17.03.21 21:49:25 (+0200) Service exited 'bluetooth sha256:a9d5faaccb7b5d33364cdf1adbeb55a07ce0ceee0cc443a13f3fe08a5f089173'
17.03.21 21:49:25 (+0200) Killing service 'airplay sha256:29679b9e1f14592d62ed8e895831eb25731c84ae1767a68da6ece9fea5b7874d'
17.03.21 21:49:25 (+0200) Service is already stopped, removing container 'airplay sha256:29679b9e1f14592d62ed8e895831eb25731c84ae1767a68da6ece9fea5b7874d'
17.03.21 21:49:25 (+0200) Killed service 'airplay sha256:29679b9e1f14592d62ed8e895831eb25731c84ae1767a68da6ece9fea5b7874d'
17.03.21 21:49:26 (+0200) Killing service 'multiroom-server sha256:23838c0d6608478de0d9085f341f85d72f316cf75ed9866c975e37285212d391'
17.03.21 21:49:26 (+0200) Service is already stopped, removing container 'multiroom-server sha256:23838c0d6608478de0d9085f341f85d72f316cf75ed9866c975e37285212d391'
17.03.21 21:49:26 (+0200) Killed service 'multiroom-server sha256:23838c0d6608478de0d9085f341f85d72f316cf75ed9866c975e37285212d391'
17.03.21 21:49:26 (+0200) Killing service 'multiroom-client sha256:c9321d0b94894c802db564253a9939bac8c8319919b8fc74cad88a4cd9e1383b'
17.03.21 21:49:26 (+0200) Service is already stopped, removing container 'multiroom-client sha256:c9321d0b94894c802db564253a9939bac8c8319919b8fc74cad88a4cd9e1383b'
17.03.21 21:49:26 (+0200) Killed service 'multiroom-client sha256:c9321d0b94894c802db564253a9939bac8c8319919b8fc74cad88a4cd9e1383b'
17.03.21 21:49:27 (+0200) Killing service 'spotify sha256:339c95b5a4bf395e051a582551f6648aa09cb4c3334a6e5fd7471aaadb53a6a9'
17.03.21 21:49:27 (+0200) Service is already stopped, removing container 'spotify sha256:339c95b5a4bf395e051a582551f6648aa09cb4c3334a6e5fd7471aaadb53a6a9'
17.03.21 21:49:27 (+0200) Killed service 'spotify sha256:339c95b5a4bf395e051a582551f6648aa09cb4c3334a6e5fd7471aaadb53a6a9'
17.03.21 21:49:27 (+0200) Killing service 'upnp sha256:35ea9dd383062abd58299fee6529df8a4df698f061735a416eb18f130a1b7294'
17.03.21 21:49:37 (+0200) Killed service 'upnp sha256:35ea9dd383062abd58299fee6529df8a4df698f061735a416eb18f130a1b7294'
17.03.21 21:49:37 (+0200) Service exited 'upnp sha256:35ea9dd383062abd58299fee6529df8a4df698f061735a416eb18f130a1b7294'
17.03.21 21:49:37 (+0200) Killing service 'sound-supervisor sha256:91b1a1ca335d6bd36779248ce199b6ed85b0313c64e6097e56da75dda0a41bd2'
17.03.21 21:49:40 (+0200) <sound-supervisor> [..................] \ : info lifecycle sound-supervisor@1.0.0~start: sound-sup

17.03.21 21:49:40 (+0200) <sound-supervisor> 
17.03.21 21:49:40 (+0200) <sound-supervisor>    ╭───────────────────────────────────────────────────────────────╮
17.03.21 21:49:40 (+0200) <sound-supervisor>    │                                                               │
17.03.21 21:49:40 (+0200) <sound-supervisor>    │      New major version of npm available! 6.14.8 -> 7.6.3      │
17.03.21 21:49:40 (+0200) <sound-supervisor>    │   Changelog: https://github.com/npm/cli/releases/tag/v7.6.3   │
17.03.21 21:49:40 (+0200) <sound-supervisor>    │               Run npm install -g npm to update!               │
17.03.21 21:49:40 (+0200) <sound-supervisor>    │                                                               │
17.03.21 21:49:40 (+0200) <sound-supervisor>    ╰───────────────────────────────────────────────────────────────╯
[... 196 lines cut ...]
24.03.21 19:55:10 (+0200)  audio  Detected audio cards:
24.03.21 19:55:10 (+0200)  audio  0 bcm2835-alsa bcm2835_alsa-bcm2835ALSA
24.03.21 19:55:10 (+0200)  audio  1 dac RPi-simple-snd_rpi_hifiberry_dac
24.03.21 19:55:11 (+0200)  audio  
24.03.21 19:55:11 (+0200)  audio  
24.03.21 19:55:27 (+0200)  audio  Waiting for sound supervisor to start
24.03.21 19:55:33 (+0200)  audio  Waiting for sound supervisor to start
24.03.21 19:55:39 (+0200)  audio  Waiting for sound supervisor to start
```

But `aplay -L` issued in audio container returns nothing:
```
root@35dd599263fc:/usr/src# aplay -L
Illegal instruction (core dumped)
root@35dd599263fc:/usr/src# aplay -l
**** List of PLAYBACK Hardware Devices ****
Illegal instruction (core dumped)
root@35dd599263fc:/usr/src#
```

Please help. What am I doing wrong? Or did I simply get bad Pimoroni DAC?

PS: [According thread](https://forums.balena.io/t/need-help-with-balena-sound-setup/273353/16) on balena forum.

PPS: I guess [this](https://github.com/balenalabs/balena-sound/issues/431) may be related.

drew-kun, 2021-03-24:
and here is **dmesg** (if it may be useful):
```
root@2e28e3c:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Sun Aug 16 09:58:55 UTC 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1e400000
[    0.000000] On node 0 totalpages: 126976
[    0.000000]   Normal zone: 1116 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x94/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:67:98:62 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=null rootfstype=ext4 rootwait vt.global_cursor_default=0 root=UUID=ba1eadef-ffdc-43e9-90fc-63f5549536d1 rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 473468K/507904K available (8055K kernel code, 683K rwdata, 2360K rodata, 8640K init, 802K bss, 26244K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
                   lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (8056 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (8640 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 684 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 803 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 28694 entries in 85 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000092] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000220] bcm2835: system timer (irq = 27)
[    0.000642] Console: colour dummy device 80x30
[    0.000721] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060374] pid_max: default: 32768 minimum: 301
[    0.060917] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060949] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062389] CPU: Testing write buffer coherency: ok
[    0.063740] Setting up static identity map for 0x8200 - 0x8238
[    0.065173] devtmpfs: initialized
[    0.076354] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.076778] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.076822] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.078192] pinctrl core: initialized pinctrl subsystem
[    0.079805] NET: Registered protocol family 16
[    0.083280] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.091086] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.091118] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.091268] Serial: AMBA PL011 UART driver
[    0.094776] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.110700] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-06-01 13:25, variant start_cd
[    0.120716] raspberrypi-firmware soc:firmware: Firmware hash is 6379679d1ec6a8c746d7e77e015f5b56b939976f
[    0.174541] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.177180] SCSI subsystem initialized
[    0.177476] usbcore: registered new interface driver usbfs
[    0.177597] usbcore: registered new interface driver hub
[    0.177855] usbcore: registered new device driver usb
[    0.180246] clocksource: Switched to clocksource timer
[    0.294043] VFS: Disk quotas dquot_6.6.0
[    0.294200] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.294519] FS-Cache: Loaded
[    0.294869] CacheFiles: Loaded
[    0.314608] NET: Registered protocol family 2
[    0.315948] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[... 362 lines cut ...]
[166899.777799] br-a6ad853699bd: port 1(vethc4caa54) entered blocking state
[166899.777821] br-a6ad853699bd: port 1(vethc4caa54) entered forwarding state
[166932.079562] eth0: renamed from vethfd6043f
[166932.172993] IPv6: ADDRCONF(NETDEV_CHANGE): vethb7aa981: link becomes ready
[166932.173289] br-a6ad853699bd: port 3(vethb7aa981) entered blocking state
[166932.173310] br-a6ad853699bd: port 3(vethb7aa981) entered forwarding state
[166936.443486] eth0: renamed from vethffdc430
[... cut ...]
[167038.418761] br-a6ad853699bd: port 2(veth8b0cf81) entered disabled state
[167038.425063] vethffdc430: renamed from eth0
[167038.637786] br-a6ad853699bd: port 1(vethc4caa54) entered disabled state
[167038.664477] device vethc4caa54 left promiscuous mode
[167038.664514] br-a6ad853699bd: port 1(vethc4caa54) entered disabled state
[167038.691033] br-a6ad853699bd: port 2(veth8b0cf81) entered disabled state
[167038.722570] device veth8b0cf81 left promiscuous mode
[167038.722606] br-a6ad853699bd: port 2(veth8b0cf81) entered disabled state
root@2e28e3c:~# 
```

tmigone, 2021-04-20:
Hi @drew-kun, this was a bug that has been fixed now (see https://github.com/balenablocks/audio/issues/62 and https://github.com/balenalabs/balena-sound/issues/429).

To fix it you'll need to re-deploy balenaSound to your device, you can do so by clicking the "Deploy with balena" button on [this](https://github.com/balenalabs/balena-sound) page. Make sure you deploy it to your existing balenaSound application where your device is and not to a newly created application.

closed 2021-04-20 by tmigone

saveriogzz, 2021-05-01:
> Hi @drew-kun, this was a bug that has been fixed now (see [balenablocks/audio#62](https://github.com/balenablocks/audio/issues/62) and #429).
> 
> To fix it you'll need to re-deploy balenaSound to your device, you can do so by clicking the "Deploy with balena" button on [this](https://github.com/balenalabs/balena-sound) page. Make sure you deploy it to your existing balenaSound application where your device is and not to a newly created application.

Hey @tmigone, this didn't unfortunately solve the issue for me! The audio keeps not to come out and I can't see anything wrong from the logs. Just the multiroom services are in the `Exited` state..

BalenaOS version: balenaOS 2.54.2+rev1
Supervisor version: 11.12.4

Could this be a problem of the pimoroni DAC itself..? Thanks