iotsound/iotsound #434: docs: add re-deploy instructions

tmigone, 2021-04-01:
A very common scenario is for users to need to re-deploy balenaSound to get latest version (for a bugfix or new feature). Would be nice for support too to have a concise explanation we can refer to.

eiddor, 2021-04-01:
@tmigone 

I'm always happy to help with docs.

When you say "re-deploy" do you just mean the same download, unzip, balena login, balana push flow that is used for install, or is there another method I don't know about?

tmigone, 2021-04-01:
hey @eiddor :wave:

Yep! Usually a user finds a problem, we fix it, and they now need to update their app to the latest version. In the forums I usually type the same wall of text saying "You need to re-deploy using the same method you originally used to deploy... yada yada". Followed by a list of steps which sometimes is detailed sometimes not.

There are a couple alternatives when deploying / re-deploying:
- The `balena push` [workflow](https://www.balena.io/docs/learn/deploy/deployment/#balena-push) where you download code as ZIP or clone using git, then use the CLI (`balena login`, `balena push`)
- The [DWB](https://www.balena.io/docs/learn/deploy/deploy-with-balena-button/) (Deploy with balena) method. It can be found in: project's readme file, https://sound.balenalabs.io/, https://hub.balena.io/

There is actually a third option, the `git push` [alternative](https://www.balena.io/docs/learn/deploy/deployment/#git-push) but thats deprecated and should not be used for new users, so we can ignore that.

So the end goal is to have it someplace in the docs where we can easily link ppl there so we don't have to type the same thing many times and also the instructions are clear and concise.

Now that I type this down, I'm not sure if we should add this to balenaSound docs or the balena docs 🤔 @vipulgupta2048 what do you think?

eiddor, 2021-04-01:
I see we have this [here](https://sound.balenalabs.io/docs/] - Do you think it's worth breaking out and putting it into the project page?

I came to this project from the balena Pi-hole project, so I always had to go look at an early balena Pi-hole blogpost every time I wanted to remember how to push.

vipulgupta2048, 2021-04-02:
Not sure how much value a redeploy part might help in the core docs where balena push is documented. This is because even when we pushing the same code again, it doesn't technically mean a redeploy. We create a new release on every push. This might end up confusing users which actually needed a redeploy but got just another push instead.

On the other hand, this makes more sense in balenaSound under the heading "Upgrading to a newer version" maybe. Where folks can be pointed to whenever they are being asked to upgrade to a newer version regardless if the instructions are exactly similar as it helps resolve the confusion of "How to upgrade to a newer version?" specifically. Let me know, I could be wrong and maybe a redeploy part in [Deployment docs](https://www.balena.io/docs/learn/deploy/deployment) adds value. Happy to help!

eiddor, 2021-04-03:
+1 @vipulgupta2048 

I'd be a fan of an "Upgrading to a newer version" section of the Getting Started page of the balenaSound docs, even if they're virtually identical to the initial install instructions.  I'm happy to work on that this weekend.

That said, I'd also see value in a "full redeploy" section in the balena Deployment docs (if it's not there already), since this would be a more advanced process and mostly used for troubleshooting/testing.  That would be helpful for me as a user and save me from creating new applications just for this purpose.

vipulgupta2048, 2021-04-05:
Looking forward to seeing what you come up with for balenaSound docs!

@eiddor Can you expand on the "Full Deploy" section that you mention on an [issue on the docs](https://github.com/balena-io/docs/issues) maybe? You can mention what the section would entail? Things you would like to see in it and how it would bring value to you as a user. We would love to help.

eiddor, 2021-04-05:
> Looking forward to seeing what you come up with for balenaSound docs!

Assuming @tmigone approves, I'll work something up this week or weekend.

> @eiddor Can you expand on the "Full Deploy" section that you mention on an issue on the docs maybe?

Will do!  Let me think through some of the use-cases and I'll add an issue.  For me, it's either about troubleshooting an issue or cleaning-up leftover files on the host after troubleshooting.  Purge Data only gets me so far.