iotsound/iotsound #435: upnp: missing plugin

tmigone, 2021-04-01:
**Describe the problem**
```
The app finds my balena-sound host but when I press Play, there is no sound and the Play button does not change to Pause (as it does when I choose the 'local renderer')...
```

**Error log**

```
29.03.21 18:13:01 (+0300)  upnp  Starting UPnP plugin...
29.03.21 18:13:01 (+0300)  upnp  Device name: cipherpunk-hifi-office
29.03.21 18:13:01 (+0300)  upnp  Using upnp UUID : 89fec2ab819f905da22037cfbee4765d
29.03.21 18:13:41 (+0300)  upnp  gmediarender 0.0.7-git started [ gmediarender 0.0.7-git (libupnp-1.8.4; glib-2.58.1; gstreamer-1.14.4) ].
29.03.21 18:13:41 (+0300)  upnp  Logging switched off. Enable with --logfile=<filename> (or --logfile=stdout for console)
29.03.21 18:14:06 (+0300)  upnp  Ready for rendering.
29.03.21 18:15:11 (+0300)  upnp  ERROR [2021-03-29 15:15:11.635989 | gstreamer] uridecodebin1: Error: Your GStreamer installation is missing a plug-in. (Debug: gsturidecodebin.c(988): no_more_pads_full (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin1:
29.03.21 18:15:11 (+0300)  upnp  no suitable plugins found:
29.03.21 18:15:11 (+0300)  upnp  gstdecodebin2.c(4640): gst_decode_bin_expose (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin1/GstDecodeBin:decodebin1:
29.03.21 18:15:11 (+0300)  upnp  no suitable plugins found:
29.03.21 18:15:11 (+0300)  upnp  Missing decoder: MPEG-4 AAC (audio/mpeg, mpegversion=(int)4, framed=(boolean)true, stream-format=(string)raw, level=(string)2, base-profile=(string)lc, profile=(string)lc, codec_data=(buffer)12100000000000000000000000000000, rate=(int)44100, channels=(int)2)
29.03.21 18:15:11 (+0300)  upnp  )
29.03.21 18:16:06 (+0300)  upnp  ERROR [2021-03-29 15:16:06.088873 | gstreamer] uridecodebin1: Error: Your GStreamer installation is missing a plug-in. (Debug: gsturidecodebin.c(988): no_more_pads_full (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin1:
29.03.21 18:16:06 (+0300)  upnp  no suitable plugins found:
29.03.21 18:16:06 (+0300)  upnp  gstdecodebin2.c(4640): gst_decode_bin_expose (): /GstPlayBin:play/GstURIDecodeBin:uridecodebin1/GstDecodeBin:decodebin1:
29.03.21 18:16:06 (+0300)  upnp  no suitable plugins found:
29.03.21 18:16:06 (+0300)  upnp  Missing decoder: MPEG-4 AAC (audio/mpeg, mpegversion=(int)4, framed=(boolean)true, stream-format=(string)raw, level=(string)2, base-profile=(string)lc, profile=(string)lc, codec_data=(buffer)12100000000000000000000000000000, rate=(int)44100, channels=(int)2)
29.03.21 18:16:06 (+0300)  upnp  )
```

tmigone, 2021-04-01:
Potential solution here: https://askubuntu.com/questions/1293814/mpeg-4-aac-decoder-and-the-h-264-decoder

drew-kun, 2021-04-02:
Hi @tmigone! Thanks for helping! Here is similar issue I've found: https://github.com/balenalabs/balena-sound/issues/342

It also kinda suggests a solution, which worked for me (please read there)

drew-kun, 2021-04-02:
basically after updating FROM stanza at `plugins/upnp/Dockerfile.template` as follows:
```
FROM balenalib/%%BALENA_MACHINE_NAME%%-debian:bullseye
```
and rebuilding the project from scratch, the issue is gone, but the stability of the UPnP sound quality is poor: the music often sticks, stops etc...

The content I try is audio, shared from my youtube music... But for that to be played, I enable "FFmpeg audio decoding for Audio formats not natively supported by renderer" in my Android BubbleUPnP app for my balena-sound render.