iotsound/iotsound #439: [WIP] DAC request master list

tmigone, 2021-04-19:
This is work in progress, please disregard it for now.
---
**Note: please read below for general guideline on wether or not you should create a new GitHub issue**

Broadly speaking for a DAC to be officially supported by balenaSound two things need to happen:
1) Drivers for the DAC must be available on balenaOS
2) DAC should be tested to work with balenaSound

balenaSound supports a lot of different DACs however it's not possible for us to test all of them individually (it would take an insane amount of time and someone with a lot of spare DACs!). For this reason we rely mostly on community contributions to help catalogue existing DACs into three categories. 

**[Supported DACs](#Supported-DACs)**
These have been tested and are known to work with balenaSound. To use them please follow instructions [here](https://sound.balenalabs.io/docs/audio-interfaces#dac-boards). If you have trouble setting up your DAC please use our public forums (https://forums.balena.io/), we'll gladly help you troubleshoot the issue. Please do not open new GitHub issues for supported DACs unless you've been instructed by our forums support agents.

**[Untested DACs](#Untested-DACs)**
This includes DACs whose drivers are included in balenaOS but haven't been tested with balenaSound most likely because no balena team member has one. If you have one of these DACs and want to help us test it please post on our forums (https://forums.balena.io/) and we'll get back to you with instructions.

**Unsupported DACs**: drivers are not available in balenaOS

### **Supported DACs**

List can be found here: https://sound.balenalabs.io/docs/audio-interfaces#dac-boards

### **Untested DACs**
- [ ] [audioinjector zero / stereo / octo / ultra / pro / isolated etc..](http://www.audioinjector.net/)
- [ ] merus audio / infineon amp #383 
- [ ] justboom-both (allows the justboom dac or amp to be used simultaneously with the digi)
- [ ] [allo boss2 / digione / katana / piano / piano plus](https://allo.com/sparky-dac.html)
- [ ] [IQaudIO codec zero](https://www.raspberrypi.org/products/iqaudio-codec-zero/)
- [ ] [hifiberry dacplusadc / adcpro /dsp / hd](https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/)
- [ ] [pifi 40](https://pifi.bz/product/pifi-amp-240/)
- [ ] [Fe-pi dac](https://www.tindie.com/products/kuupaz/fe-pi-audio-z-v2/)
- [ ] [apple-pi / pecan pi dac](https://orchardaudio.com/pecanpi)
- [ ] [audiosense pi](https://gitlab.com/kakar0t/audiosense-pi)
- [ ] [Audiophonics I-Sabre ES9028Q2M](https://www.audiophonics.fr/en/dac-and-interfaces-for-raspberry-pi/audiophonics-i-sabre-dac-es9028q2m-raspberry-pi-3-2-a-b-spdif-i2s-p-12105.html)
- [ ] [terraberry dac](https://www.terratechnos.com/products/terraberrydac/index-e.html)
- [ ] [various suptronics dacs](http://www.suptronics.com/boards.html)
- [ ] [super audio board](https://github.com/whollender/SuperAudioBoard)
- [ ] [bassfly / bassfly-mic](https://github.com/Darmur/bassfly-uhat)
- [ ] [max98357a](https://www.adafruit.com/product/3006)
- [ ] pibell - can't find link for this
- [ ] mbed-dac - can't find link for this
- [ ] [audu7002-simple](https://www.analog.com/en/products/adau7002.html#)
- [ ] [picap](https://www.bareconductive.com/shop/pi-cap/)
- [ ] [Google AIY voiceHAT v1/v2](https://forums.balena.io/t/almost-got-google-aiy-dac-working/206819/13)
- [ ] [phat dac / phat beat / picade hat / speaker phat](https://shop.pimoroni.com/products/phat-dac) - these will already work but no longer in production so not a huge amount of point
- [ ] [dionauadio dacs / amps](https://dionaudio.nl/categorie/producten/streaming/)
- [ ] [raspiaudio dac range](https://raspiaudio.com/raspiaudio)
- [ ] [osa dacberry one / pro](https://www.osaelectronics.com/product/dacberry-pro/)
- [ ] [nanosound dac](https://nanomesher.com/nanosounddac/?gclid=CjwKCAiArbv_BRA8EiwAYGs23Eaol-qncGt1PYGGgKrnx-XO9JHNh8UGF395L2HBRnQR-2vrg4v43RoC5-0QAvD_BwE)
- [ ] [pi 2 design 502 / 502 pro / 503](https://www.pi2design.com/502dac.html)
- [ ] [mamboberry dac](https://collybia.com/product/mamboberry-hd-dac-192khz-24bit/)
- [ ] [HifiBox dac](https://m.aliexpress.com/item/32893484644.html)
- [ ] [generic rpi-dac](https://www.runeaudio.com/forum/generic-i-s-es9018k2m-dac-board-t6177.html)
- [ ] [rpi-cirrus-wm5102](http://www.farnell.com/datasheets/1928993.pdf) - discontinued so maybe no point
- [ ] [red rocks audio digidac1 - rra-digidac1-wm8741-audio]()

### **Unsupported DACs**
- [ ] [bassowl](https://github.com/Darmur/bassfly-uhat)
- [ ] [tau dac](http://taudac.com/)
- [ ] [raspidac3](https://www.grulich.eu/files/RaspiDACv3.pdf)
- [ ] [WM8960 Audio HAT](https://www.waveshare.com/wiki/WM8960_Audio_HAT) - #264 
- [ ] [Adafruit Voice Bonnet](https://learn.adafruit.com/adafruit-voice-bonnet/audio-setup) - I believe this is almost identical to the WM8960 from waveshare (abvoe) and the seeed respeaker 2 and keyestudio respeaker (below)
- [ ] [respeaker 2 / 4 linear / 6 circular](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/) - #166 - interestingly this [also supports jetson nano](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/#platform-supported)
- [ ] [keyestudio respeaker 2](https://wiki.keyestudio.com/Ks0314_keyestudio_ReSpeaker_2-Mic_Pi_HAT_V1.0) clone of the seeed respeaker above #166 
- [ ] [sabreberry32](https://nw-electric.way-nifty.com/blog/sb32j.html)
- [ ] [hifibunny q2m](https://www.diyaudio.com/forums/digital-line-level/314935-es9038q2m-board.html?s=03ad118356763b9d3de42435ae78f4c4)

tonydrabeck, 2021-09-16:
Please add support for the Audiobox USB 96

kbftech, 2022-10-12:
I2S Interface Pifi DAC PCM5122 DAC + HiFi DIGI Digital Audio Card Analogue Output Signal for Raspberry Pi B+ / 2 Model B / 3B (64-bit)

https://www.amazon.ca/gp/product/B08HSYTD4F/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

[image]

Quite the name, thanks Amazon! Super cheap DAC. I tried all the dt_overlay options that mentionned the chip 5122 or 512x. Although I got the soundcard recognized according to the console, there is no sound output from the RCA nor the line-out.

I bought it based on the review saying it worked with Volumio when using "hifiberry DAC plus". Although the "hifiberry-dacplus" didn't work, the "hifiberry-dac" did get the dac recognized:
[image]

No sound though. Also, I get these errors, but since I get them with my other DAC (which works) I guess it doesn't matter much:
```
 audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
 audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
 audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```

kbftech, 2022-10-13:
By the way, the other one that works is the following (If you want to add it to the list):
https://www.amazon.ca/Raspberry-Expansion-Board-PCM5122-24-bit/dp/B07MM4VP5C/ref=d_pd_sbs_sccl_4_1/147-2367878-6664069?pd_rd_w=g8qF9&content-id=amzn1.sym.e09d3028-5638-44d6-a6a2-f832afb24084&pf_rd_p=e09d3028-5638-44d6-a6a2-f832afb24084&pf_rd_r=YRCWKX75CHDPETPX6PZ4&pd_rd_wg=xrzOV&pd_rd_r=bcf59abb-d021-4e31-94c8-0dd5adc718c1&pd_rd_i=B07MM4VP5C&psc=1
Raspberry Pi Expansion Board Raspberry Pi HiFi DAC+ HD Audio PCM5122 24 bit Expansion Board for Raspberry Pi 3 Model B 2B B+ A+ Raspberry Pi Zero W

BALENA_HOST_CONFIG_dtoverlay = "allo-piano-dac-pcm512x-audio"

plisker, 2024-01-22:
Would it be possible for me to use a `Behringer UCA202`?