iotsound/iotsound #441: Noticeable latency on Spotify

halfcyan, 2021-04-19:
**Describe the problem**
There is a noticeable delay on Spotify of around 2 seconds

**Expected behavior**
I expected there to be at least less delay (around half a second max) because I have seen many Spotify Connect devices with little to no noticeable latency.

**Device information**
 - Board (hardware info): Raspberry Pi 3B+
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS 2.72.0+rev1

**Additional context**
I'm basically just asking if this is normal or if it should be less delay

halfcyan, 2021-05-04:
Just figured something out, it isn't an issue with Spotify as the song loads instantly, it's an issue with the audio stack. It's gotten worse though, it's about 14 seconds now.

eiddor, 2021-05-04:
I always get a delay, but it's never more than a couple of seconds.  Does this happen with Bluetooth, too?

npmstart-pray, 2021-05-16:
is the Pi wireless or connected via ethernet?

xvisor16, 2021-05-22:
Same problem here! When I reboot all pi"s (4) latency is like 1 sec... But after a day or 2 this runs up with a few seconds. Last week I hadn't used Balena for over a week or 2 and the total delay was beyond 15 seconds...

npmstart-pray, 2021-05-23:
I guess a reboot daily wouldn’t be a bad thing…mayhaps it could be automated?

Sent from nowhere near my ready room with my new tricorder

jswetzen, 2021-06-06:
I get the same, but with AirPlay. The delay grows over time, so I reboot daily via the API. But for me the AirPlay delay has actually been worse than it is with Spotify.

halfcyan, 2021-06-07:
It was across everything. I figured it out though, I had to disable multiroom. I think it was constantly searching for a receiver and never found one because I don't have another multiroom client.

halfcyan, 2021-06-07:
> I get the same, but with AirPlay. The delay grows over time, so I reboot daily via the API. But for me the AirPlay delay has actually been worse than it is with Spotify.

I think there is an inherent 2 second delay in the AirPlay implementation

ChestersGarage, 2021-06-24:
The input protocols introduce a chunk of latency in order to prevent playback issues when there may be network performance problems. However, there is a known issue with latency on the input slowly increasing over days of uptime, when running in multi-room mode.  After about a week, it can get to over 20 seconds.  This issue makes note of it (https://github.com/balenalabs/balena-sound/issues/354), but I was not able to find an issue that directly addresses it.  I will create one.

The workaround is to restart the multiroom-server container on the device being used as the server, or just reboot it.  This resets that latency issue back to the usual ~2 seconds.

npmstart-pray, 2021-06-24:
They’ll have to resolve this for commercial deployments. A daily reboot at home isn’t a big deal, but command latency creep at a business wouldn’t be well received. 

Sent from nowhere near my ready room with my new tricorder

denis-heka, 2021-07-17:
+1 for this issue. I'm getting a 25-30s delay when using spotify connect. Any changes to volume, start/stop make it pretty unusable

npmstart-pray, 2021-07-17:
Even with a reboot?

Sent from nowhere near my ready room with my new tricorder

DenisKent, 2021-07-18:
My bad, I thought it had been rebooted. Back to 2-3s delay after a fresh reboot. Still frustrating though. I'll look into the possibility of making the intended latency variable.

npmstart-pray, 2021-07-18:
Yeah, that’s still even too long. A few hundred ms or less would be best. 
I’m wondering if subnetting sound devices would be of any help there - my network isn’t all that busy. Perhaps someone can enlighten if they have more networking knowledge than I if that’s a viable option. 

Sent from nowhere near my ready room with my new tricorder

mularkey, 2021-12-03:
Sorry to revive this thread but has this been solved? Has anyone smarter than me written a script to reboot at, say 6 am each morning?

jswetzen, 2022-03-01:
I have a script that calls the API every night. If you have a Linux server, a `curl` call will do
> curl -s -X POST --header "Content-Type:application/json" --header "Authorization: Bearer YOUR-API-TOKEN-HERE" --data '{"uuid": "YOUR-DEVICE-UUID-HERE"}' "https://api.balena-cloud.com/supervisor/v1/reboot"

closed 2022-04-05 by tmigone