iotsound/iotsound #448: Broken link

mafhper, 2021-05-14:
**Describe the problem**
A clear and concise description of what the problem is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://sound.balenalabs.io/docs/device-support/#recommended'
2. Scroll down to 'Notes'
3. Click on 'audio interfaces'
4. See error

**Expected behavior**
Acess to the link https://sound.balenalabs.io/docs/audio-interfaces/

**Screenshots**
[image]

[image]

**Device information**
 - Windows 10 HOME 20H2 x64
 - Google Chrome version 90.0.4430.212 64 bits

eiddor, 2021-05-14:
This issue seems to exist for multiple links on multiple pages depending if the original page's URL ends in a trailing-slash or not.

For instance - Clicking "audio interfaces" on: 

`https://sound.balenalabs.io/docs/device-support/`

and

`https://sound.balenalabs.io/docs/device-support`

will produce different results (the second one works) - If you start at a page using a link in the side bar, the next link will work.

I'm not 100% familiar with how the doc system works, but I'm poking around to see if I can figure out why.

eiddor, 2021-05-14:
To add in case someone is looking at the back-end - the resulting target URLs for the two examples above are:

`https://sound.balenalabs.io/docs/docs/audio-interfaces`

and

`https://sound.balenalabs.io/docs/audio-interfaces`

The second one is accurate.  There are ways to fix this in Markdown by making the URLs non-relative, but I'm guessing the root cause is somewhere else since the docs haven't changed in a while.

tmigone, 2021-05-24:
This looks like a bug with the tool we use to autogenerate the documentation site. I've reported it here: https://github.com/product-os/landr/issues/377

The repo is private at the time so I'll paste my findings:
> When cross linking to another landr page there is something weird behaviour going on. Links wont work for the current page that’s being loaded but if I navigate to another one it does work.

> For example:

> - if I go to https://sound.balenalabs.io/docs/getting-started/ the first link below the photo has an extra /docs so it breaks.
> - if however I go to https://sound.balenalabs.io/docs/device-support and then navigate to Getting started section using the sidebar then the first link below the photo will work just fine.

eiddor, 2021-05-24:
Thanks, @tmigone! Some kind of rewrite rule with trailing-slashes, I suspect.