iotsound/iotsound #463: Airplay refuses to run - pa context error

AliMickey, 2021-06-07:
**Describe the problem**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the git
2. Adjust Dockerfile parameters for device type.
[image]
3. docker-compose up -d

Spotify works with no problems. 

**Expected behavior**
Airplay works

**Device information**
 - Board  Raspberry Pi 3
 - Device type: e.g. Raspberry Pi 3 running Raspbian

TutosJuegosFree, 2021-06-07:
I have the same problem with a raspberry pi 3b+, the only way I found to solve it is to restart the service every time the computer boots up

tmigone, 2021-11-24:
Fixed in v3.6.11. For details see https://github.com/balenalabs/balena-sound/pull/508

commit 9b32691e (2021-11-24): airplay: fix connection bug with airplay

Closes: #463, #501, #497, #379
Change-type: patch
commit e4e5f8c0 (2021-11-24): airplay: fix connection bug with airplay

Closes: #463, #501, #497, #379, #489, #483
Change-type: patch
closed 2021-12-07 by tmigone