iotsound/iotsound #464: Balena Sound and bluetooth audio transmission

Tamarozzi, 2021-06-07:
After several attempts, versions, dongles and configurations, it is very frustrating not to be able to make a technology work that was presented to me as easy. Really the entire deployment is easily accessible and hassle-free, until you want to make it work in practice. From then on, nothing else worked, playback lasts around 3 seconds, the errors presented (SBC decoding error (-2), Unsupported fragmented SBC frame) left me very unmotivated, there is nothing clear about these problems, just comments that lead us to other factors like power supply and bluetooth adapters that don't result in any solution. A feather.

JaragonCR, 2021-06-10:
Are you having this on a Pie3?

Tamarozzi, 2021-06-10:
> Você está tendo isso em um Pie3?

Sim, Raspberry Pi3 Model B V1.2

JaragonCR, 2021-06-10:
https://github.com/balenalabs/balena-sound/issues/62

Solution was to 
mount -o remount,rw /
vi /usr/bin/btuart

make this change

else
                # This is the default one
                #$HCIATTACH /dev/serial1 bcm43xx 921600 noflow - $BDADDR
                $HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR

I tested it last night and was pretty stable. There is another issue with UPNP and sound control

Tamarozzi, 2021-06-10:
I will try this change, as soon as possible I will get back to you with the result, thank you

tjoels, 2021-08-28:
> 
> 
> So I did a lot of testing today, this is the only wat to get a 3 working with Bluetooth.
> 
> Use balenaOS 2.38.0+rev1
> 
> Once deployement is complete
> mount -o remount,rw /
> vi /usr/bin/btuart
> 
> make this change
> 
> else
> # This is the default one
> 
> #$HCIATTACH /dev/serial1 bcm43xx 921600 noflow - $BDADDR
> $HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
> ---Save File----
> systemctl edit --full bluetooth
> 
> Add --Plugin like below
> 
> ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp
> 
> ---Save File----
> 
> Reboot

This trick does not seem to work in balenaOS 2.80.3+rev1, supervisor version 12.7.0 on a Raspberry Pi 3B+.

JaragonCR, 2021-09-02:
> > So I did a lot of testing today, this is the only wat to get a 3 working with Bluetooth.
> > Use balenaOS 2.38.0+rev1
> > Once deployement is complete
> > mount -o remount,rw /
> > vi /usr/bin/btuart
> > make this change
> > else
> > # This is the default one
> > #$HCIATTACH /dev/serial1 bcm43xx 921600 noflow - $BDADDR
> > $HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
> > ---Save File----
> > systemctl edit --full bluetooth
> > Add --Plugin like below
> > ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp
> > ---Save File----
> > Reboot
> 
> This trick does not seem to work in balenaOS 2.80.3+rev1, supervisor version 12.7.0 on a Raspberry Pi 3 Model B.

I abandoned my 3 because BT was unreliable, those changes do work, but once you are in a 4 you realize there's no fix like HW.

closed 2022-04-05 by tmigone