iotsound/iotsound #477: Input latency slowly increases over days in multi-room mode

ChestersGarage, 2021-06-24:
**Describe the problem**
I thought we had already created an issue for this, but I couldn't find anything for it.  It has been discussed in the following issues:
https://github.com/balenalabs/balena-sound/issues/333
https://github.com/balenalabs/balena-sound/issues/294
https://github.com/balenalabs/balena-sound/issues/354

Playback latency slowly creeps upward over days of server device uptime.  After about a week, the latency will be nearly 20 seconds. 

**To Reproduce**
Allow a fleet to remain operational without changes or restarts to application containers or the devices for several days.  Latency will be noticeably higher after a day or two.

**Expected behavior**
Playback latency should remain consistent (~2 seconds).

**Device information**
 - Board (hardware info): Raspberry Pi 4B+
 - Device type (as seen in balenaCloud): Raspberry Pi 4 (using 64bit OS)
 - balenaOS version: balenaOS 2.67.3+rev4

influenist, 2021-07-09:
I have the same. With just one device "lagg" increases by the date.

kbftech, 2022-11-03:
I can concur that this affects my system very much. It takes a long time before any change actually happens (volume adjustments, song swap). I'm using two raspberry pi 3B with latest version available at this time (afaik):
[balenaOS 2.105.1+rev1](https://github.com/balena-os/meta-balena/blob/master/CHANGELOG.md)

jaskaransarkaria, 2023-09-16:
OS: [balenaOS 3.1.12](https://github.com/balena-os/meta-balena/blob/master/CHANGELOG.md)
System: Raspberry Pi 4
Notes: Bluetooth enabled

Current workaround:
[image]