iotsound/iotsound #478: Persistent storage for snap server data

ChestersGarage, 2021-06-24:
**Is your feature request related to a problem? Please describe.**
I run the Snap.Net system tray applet on my Windows computer and Android phone, both of which allows me to manage device volumes and give names to each device on my fleet.  However, each time the snapcast server is restarted, either because the multiroom-server container is restarted or the device is rebooted, the information I've set for my fleet is lost.  

**Describe the solution you'd like**
There appears to be a way to persistently store the device info on the snapcast server, using a configuration option in snapserver.conf.

[server]
datadir = /var/cache/snapcast/

And then provide that location as a volume in Docker.  (My local repo is too messed up to try this myself at the moment.)  

Configured Snap.Net client: 
[image]

Initial state, and what it looks like any time I restart the sanpserver:
[image]

jannis3005, 2021-08-06:
I'd love to see this too. I've been trying to use homeassistant to do some automations, but everytime one or all of the devices are restartet, the group names change and thus breaking all set automations.

salsabeard, 2021-09-16:
> I'd love to see this too. I've been trying to use homeassistant to do some automations, but everytime one or all of the devices are restartet, the group names change and thus breaking all set automations.

I've been struggling with this exact problem as well.  Between Snapcast and Plex, I had to kill of media player discovery entirely because it has just been adding WAY too many dynamic entities.

PHLAK, 2025-04-07:
I'm pretty sure I just ran into this issue after updating my (custom) Docker container image. Has there been any progress on this?

Where is this data stored normally? I don't think it's in memory since restarting the container doesn't cause issues. But where on disk is this stored?