iotsound/iotsound #483: Airplay doesn't show on Apple devices on initial boot up.

magauthority, 2021-07-26:
**Describe the problem**
When booting my rp3b+ airplay service starts and runs fine, but doesn't show up on any devices on the local wifi. A restart of the airplay service corrects the issue. Happens on both my balena-sound Rpi devices.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on pi.
2. Wait for it to boot and be 'ready'
3. Open another apple device (iphone, ect) and look for device listing on airplay - not there. 
4. open balena sound app, and restart airplay service, then device shows up and works at expected. 

**Expected behavior**
Airplay should be available upon initial startup.

**Device information**
 - Board (hardware info)  Raspberry Pi 3B+
 - Device type (as seen in balenaCloud): [e.g. Raspberry Pi 3 (using 64bit OS)]
 - balenaOS version: balenaOS 2.80.3+rev1

**Additional context**
Add any other context about the problem here.

pjmartorell, 2021-07-30:
I have the same issue in a RP3 model B with a fresh installation. After rebooting the airplay service the device appeared, but not at first.

thewanuza, 2021-09-28:
I have the same problem, I must restart airplay service for see rpi on apple device.

airplay  fatal error: pa context is not good -- the error message "Connection terminated"

tmigone, 2021-11-24:
Fixed in v3.6.11. For details see https://github.com/balenalabs/balena-sound/pull/508

scerazy, 2023-02-25:
Same happens on clean install on RPi3B with 3.9.6