iotsound/iotsound #484: Raspberry Pi 4 (using 64bit OS) issues - localhost', error: Connection refused

consurity, 2021-07-30:
Having issues with various services on balena-sound. 

Already tried to re-deploy Balena-sound, as well as running Balena-OS-DEV to no avail. 

Bluetooth is available, as well as spotify, but while playing audio through each of these channels, I don't get any output on the Pi. 

**OS** : balenaOS 2.82.10+rev1 - development
**Device** :  Raspberry Pi 4 (using 64bit OS)
**Logs:** 

```
30.07.21 20:05:56 (+0200) <multiroom-client> 2021-07-30 18-05-56.144 [Error] (Connection) Error in socket shutdown: Socket not connected
30.07.21 20:05:57 (+0200) <multiroom-client> 2021-07-30 18-05-57.146 [Error] (Connection) Failed to connect to host 'localhost', error: Connection refused
30.07.21 20:05:57 (+0200) <multiroom-client> 2021-07-30 18-05-57.146 [Error] (Controller) Error: Connection refused
30.07.21 20:05:57 (+0200) <multiroom-client> 2021-07-30 18-05-57.146 [Error] (Connection) Error in socket shutdown: Socket not connected
30.07.21 20:05:58 (+0200) <multiroom-client> 2021-07-30 18-05-58.147 [Error] (Connection) Failed to connect to host 'localhost', error: Connection refused
30.07.21 20:05:58 (+0200) <multiroom-client> 2021-07-30 18-05-58.147 [Error] (Controller) Error: Connection refused
30.07.21 20:05:58 (+0200) <multiroom-client> 2021-07-30 18-05-58.147 [Error] (Connection) Error in socket shutdown: Socket not connected
30.07.21 20:05:58 (+0200) <airplay> Waiting for audioblock to start...
```

[image]

[image]

aaronmaziarz, 2021-08-07:
I am having a very similar (likely the same) issue.  I have made small troubleshooting efforts including updating the OS and supervisor, stopping bluetooth and spoify plug-in services, and attempting to backtrack the issue in the logs.  In addition to the errors consurity included above, I found some logs from the audio block package which could be of interest in solving the issue.   

OS : balenaOS 2.82.10+rev1 - development
Device : Raspberry Pi 4 (using 64bit OS)
Supervisor Version : 12.10.1
Logs:
[balena-sound_poolside @ vσνice-06.08.21_18_23_32_(-0700).txt](https://github.com/balenalabs/balena-sound/files/6948400/balena-sound_poolside.%40.v.ice-06.08.21_18_23_32_.-0700.txt)

Thank you for your help and all you guys do to make Balena Sound such a great open source software program :)

[image]

[image]

tmigone, 2022-04-05:
Hey can you try reproduce this with latest balenaSound version 3.8.9? Feel free to re-open this issue if the problem persists. Thanks!

closed 2022-04-05 by tmigone