iotsound/iotsound #486: I can't install tails with BalenaEtcher

Ryan5013, 2021-08-09:
Every time I select tails then select target and then hit flash it asks for a password. it says balenaEtcher wants to make changes. Type your password to allow this. What is the password for this to work???

eiddor, 2021-08-09:
I'm not sure what tails is, but when belenaEtcher asks for a password it usually wants the password on your system (for example, if you're running it on a Mac, you enter your Mac user's password).

Ryan5013, 2021-08-09:
I tried that but it did not work

eiddor, 2021-08-09:
Does your account have administrator access?  It's hard to tell without know what kind of system you're running on, but maybe you need to run it as administrator (I think that's a Windows thing).

Ryan5013, 2021-08-09:
I am using a iMac

Ryan5013, 2021-08-09:
software version Mac OS Big Sur 11.5.1

eiddor, 2021-08-09:
On my Mac, I just enter my regular Mac password.

See this thread: https://github.com/balena-io/etcher/issues/2997

eiddor, 2021-08-09:
Just a note - This repo is for balenaSound.

Ryan5013, 2021-08-09:
THANKS FOR THE HELP I WILL HAVE TO FIGURE IT OUT IF THERE IS NOTHING ELSE YOU KNOW TO HELP

closed 2022-01-24 by tmigone