iotsound/iotsound #493: USB soundcards not being recognized

spenserkm, 2021-09-08:
**Describe the problem**
RPi4 running Balena does not produce audio over USB soundcards

**To Reproduce**
Boot RPi with Balena with USB soundcards plugged in -- audio will play over the 3.5mm but not the USB soundcard

**Expected behavior**
RPi should default to playing over USB, per the Balena docs

**Device information**
I have a RPi4 4gb that is running balenaOS 2.83.10+rev1 and is on the 3.6.9 balena-sound release.

Spotify Connect works fine, and I can get audio from the onboard 3.5mm. However, even when I reboot with a USB soundcard plugged in (I have tried both an SMSL SU-9 and an Audioengine D1), it does not find the USB soundcard.

I have not changed DT overlay, since I have read that versions 3.0 and up should not require any changes to variables and should default to USB sound on boot up.

alsamixer does not show the USB soundcards as an output option.

Here’s what I get when I list sinks:

```
bash-5.0# pactl list sinks
Sink #0
State: RUNNING
Name: alsa_output.bcm2835-jack.stereo-fallback
Description: Built-in Audio Stereo
Driver: module-alsa-card.c
Sample Specification: s16le 2ch 44100Hz
Channel Map: front-left,front-right
Owner Module: 6
Mute: no
Volume: front-left: 49152 / 75% / -7.50 dB, front-right: 49152 / 75% / -7.50 dB
balance 0.00
Base Volume: 56210 / 86% / -4.00 dB
Monitor Source: alsa_output.bcm2835-jack.stereo-fallback.monitor
Latency: 64705 usec, configured 66666 usec
Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY
Properties:
alsa.resolution_bits = “16”
device.api = “alsa”
device.class = “sound”
alsa.class = “generic”
alsa.subclass = “generic-mix”
alsa.name = “bcm2835 Headphones”
alsa.id = “bcm2835 Headphones”
alsa.subdevice = “0”
alsa.subdevice_name = “subdevice #0”
alsa.device = “0”
alsa.card = “0”
alsa.card_name = “bcm2835 Headphones”
alsa.long_card_name = “bcm2835 Headphones”
alsa.driver_name = “snd_bcm2835”
device.bus_path = “/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0”
sysfs.path = “/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0”
device.form_factor = “internal”
device.string = “hw:0”
device.buffering.buffer_size = “131072”
device.buffering.fragment_size = “131072”
device.access_mode = “mmap+timer”
device.profile.name = “stereo-fallback”
device.profile.description = “Stereo”
device.description = “Built-in Audio Stereo”
alsa.mixer_name = “Broadcom Mixer”
module-udev-detect.discovered = “1”
device.icon_name = “audio-card”
Ports:
analog-output-headphones: Headphones (priority: 9900)
Active Port: analog-output-headphones
Formats:
pcm

Sink #1
State: RUNNING
Name: balena-sound.input
Description: Null Output
Driver: module-null-sink.c
Sample Specification: s16le 2ch 44100Hz
Channel Map: front-left,front-right
Owner Module: 23
Mute: no
Volume: front-left: 49152 / 75% / -7.50 dB, front-right: 49152 / 75% / -7.50 dB
balance 0.00
Base Volume: 65536 / 100% / 0.00 dB
Monitor Source: balena-sound.input.monitor
Latency: 10808 usec, configured 66666 usec
Flags: DECIBEL_VOLUME LATENCY
Properties:
device.description = “Null Output”
device.class = “abstract”
device.icon_name = “audio-card”
Formats:
pcm

Sink #2
State: RUNNING
Name: balena-sound.output
Description: Null Output
Driver: module-null-sink.c
Sample Specification: s16le 2ch 44100Hz
Channel Map: front-left,front-right
Owner Module: 24
Mute: no
Volume: front-left: 65536 / 100% / 0.00 dB, front-right: 65536 / 100% / 0.00 dB
balance 0.00
Base Volume: 65536 / 100% / 0.00 dB
Monitor Source: balena-sound.output.monitor
Latency: 3831 usec, configured 20000 usec
Flags: DECIBEL_VOLUME LATENCY
Properties:
device.description = “Null Output”
device.class = “abstract”
device.icon_name = “audio-card”
Formats:
pcm

Sink #3
State: RUNNING
Name: snapcast
Description: Null Output
Driver: module-null-sink.c
Sample Specification: s16le 2ch 44100Hz
Channel Map: front-left,front-right
Owner Module: 25
Mute: no
Volume: front-left: 65536 / 100% / 0.00 dB, front-right: 65536 / 100% / 0.00 dB
balance 0.00
Base Volume: 65536 / 100% / 0.00 dB
Monitor Source: snapcast.monitor
Latency: 6945 usec, configured 23219 usec
Flags: DECIBEL_VOLUME LATENCY
Properties:
device.description = “Null Output”
device.class = “abstract”
device.icon_name = “audio-card”
Formats:

```
Any help would be much appreciated! Thanks all.

Zageron, 2021-09-09:
I am seeing the same behaviour after browsing the forum for solutions.
I am using the Schiit Modi 2/3. Neither show up as usb devices in audio or bluetooth audio.

spenserkm, 2021-09-12:
I had to switch to Volumio -- I'll revisit Balena in a few months because it's so much better from a devops perspective than volumio.

Zageron, 2021-09-22:
> I had to switch to Volumio -- I'll revisit Balena in a few months because it's so much better from a devops perspective than volumio.

How is volumio treating you? I'm still looking for a good solution.

Zageron, 2021-09-25:
Tried volumio and had zero luck getting it to work. Usb audio works out of the box but the thing barely connects to Spotify. :(

Zageron, 2021-10-18:
I'm now using moode.