iotsound/iotsound #495: Dedicated Server as a Docker Container

salsabeard, 2021-09-16:
**Is your feature request related to a problem? Please describe.**
One of the absolute biggest issues faced with multi-room setups is resource availability of the devices to act as a server and a client simultaneously.

**Describe the solution you'd like**
I'd like to have an option to properly/ officially deploy Balena Sound as a standalone docker container that can act as a dedicated server for other devices.  My thought is that the implementation would be a stripped down variant that is built to handle supervisor and sound input tasks.

**Describe alternatives you've considered**
I've considered purchasing another raspberry pi to use as a standalone server, but I already have a lot of various containers running on my Synology and would love to have the option to harness its resources for my sound system as well.

**Additional context**
I've also considered that this dedicated server could have a lightweight Web UI integrated to allow users to have a very simple way of controlling volume for the multi-room clients, but also that it could have a way to set and unset clients that are participating in the playback group.

akifreak, 2021-10-06:
I would love to have one master instance running on a powerful server and then having several Raspberry Pi Zeros running in Multi Room Client mode. This would be an extremly cost effective solution !

jannis3005, 2022-01-25:
This would also give the option to use balena sound along side other services. i have one raspberry pi running homeassistant and some other things. i would love to just use it as a multiroom-server instead of buying another pi.