iotsound/iotsound #496: USB DAC isn't recognized (SONY UDA-1)

buzuddha, 2021-09-19:
Hi all,

I'm working with a SONY UDA-1 connected via USB. I've tried power cycling the pi but I'm still getting audio through the headphones.

Anything I can do to force USB audio?

Thanks!

HOST OS VERSION
balenaOS 2.83.10+rev1
Raspberry Pi 4 (using 64bit OS)Raspberry Pi 4 (using 64bit OS)
CURRENT RELEASE
d2a9ae8
SUPERVISOR VERSION
12.9.3

joegibes, 2021-09-27:
I don't have that DAC but I had this problem getting my Focusrite Scarlett 2i4 to work over USB.

This solution worked for me: https://github.com/balenalabs/balena-sound/issues/314#issuecomment-715248194
@cpwood
> FWIW, I was able to workaround this temporarily by getting an SSH session in the audio service and issuing the following commands:
> 
> `pactl list`
> 
> to find the name of the sink for my USB DAC, and then using that name:
> 
> `pactl set-default-sink alsa_output.usb-soundcard-1.analog-stereo`

I actually was able to set Device Variable (in balena dashboard) `AUDIO_OUTPUT` = `alsa_output.usb-soundcard-0.analog-surround-40` (the sink I found from `pactl list`). 

I theorize that I had this problem (even on the latest balenaSound) because the Scarlett 2i4 has 4 outputs - thus making it 'surround' rather than 'stereo'. Maybe the Sony has multiple outputs too?

buzuddha, 2021-09-30:
Do I need to install pactl?
```

root@balena:~# pactl list
bash: pactl: command not found
```

eiddor, 2021-09-30:
You have to be in the audio container - you will find `pactl` there.

buzuddha, 2021-10-02:
Hmmm `pactl list | grep usb` doesn't  produce any result, even after rebooting the pi with the dac plugged in and turned on.

plisker, 2024-02-02:
> I actually was able to set Device Variable (in balena dashboard) AUDIO_OUTPUT = alsa_output.usb-soundcard-0.analog-surround-40 (the sink I found from pactl list).

Was going crazy, this worked for me as well with my own USB device. Thanks!