iotsound/iotsound #499: Sound supervisor doesn't start

tkraus13, 2021-10-02:
The overall problem: My device no longer allows bluetooth connections, and I suspect it is because the Sound Supervisor service refuses to start. 

On first install, I was able to connect to the balena-sound device through bluetooth, but it would not play sound through the audio jack. After doing some reading I thought it was because I needed to include the drivers for the specific audio dac (Raspiaudio Audio+) so I followed the directions here (https://sound.balenalabs.io/docs/audio-interfaces/) and after performing the power cycle I was no longer able to connect to the device through bluetooth. I have tried restarting it several times, removing it from the list of bluetooth devices on my phone, and restarting the individual services, and the only service that doesn't start is Sound Supervisor, which makes me think that is the problem. And on top of this I haven't yet been able to verify if I can play audio through it. 

Hardware: 
Raspberry Pi Zero W
Raspiaudio Audio+ DAC
balenaOS 2.83.21
**Describe the problem**
A clear and concise description of what the problem is.

tkraus13, 2021-12-01:
An update: I got BalenaSound working on the Raspberry Pi 3 and was able to use to play sound over bluetooth several times, even through multiple reboots. However, now I am having the same issue as before in that I am no longer able to connect to the Raspberry Pi over bluetooth. I have not changed anything in the OS since I originally installed it, nor have I done anything different in my startup and shutdown sequences in regards to the Pi. It was working perfectly yesterday, and now it's not. Any help would be appreciated, however since something that is marketed to be easy to use and setup is in reality useless and extremely difficult to even attempt to setup and work, I'll probably be moving on to other services to try to get this working.

npmstart-pray, 2021-12-01:
I'm increasingly frustrated at the supervisor not starting as well. I've tried re-flashing current and older versions of balenaOS, both dev and production branches...nada. And what really burns me is that this is the 3rd instance I'm trying to get going since May of this year, the 2nd happening in early July, and they had no problem. Now with whatever change to balenacloud, something breaks in this regard. @tmigone has said starting this month he'll turn his attention to balena sound, and I hope he's at it in earnest.