iotsound/iotsound #500: No Services

Rickexe, 2021-10-04:
Hi,

[image]

I have rebuild the image countless times. It doesn't show services. I am on the latest version raspberry pi 3 as you can see. I don't know how to make this work.

eiddor, 2021-10-05:
I can't really tell from the screenshot, but I assume this device is part of the balena-sound application.

Do you get any errors when you push the application?

It might be worth doing a "Purge Data" (the pulldown beside the light bulb).

Rickexe, 2021-10-05:
What do you mean by pushing the application?

eiddor, 2021-10-05:
Generally, you "push" an image to your balena fleet (formerly known as application), either at the CLI or using the one-click deploy.  Do you get any errors or messages when you do that?

How exactly are you building the image?  What steps have you taken so far?

Rickexe, 2021-10-05:
I built the image for raspberry pi 3 and flashed those to sd cards. I did the same thing couple months back and it work flawlessly. Now I did the exact same thing I don't see any services on neither of the pi's. 

I can't find "push" anywhere in the balena deploy thing. It's not like github, is it?

eiddor, 2021-10-05:
So after you build the image and boot from the cards, you have to deploy the application/fleet either using the one-click deploy or the balena CLI:

https://sound.balenalabs.io/docs/getting-started

eiddor, 2021-10-07:
All balena projects need to be deployed/pushed after you create the device image and fleet, so you may have used "one click deploy" in the past maybe?  

Let us know how it works out.

tmigone, 2022-01-24:
Hey @Rickexe closing this due to inactivity, let us know if you need more help. Cheers!

closed 2022-01-24 by tmigone