iotsound/iotsound #511: Balena Sound RPi4 - no bluetooth, airplay or spotify

Getafix17, 2021-12-04:
Hello I cannot see my Balena sound on Bluetooth or Spotify (I have a premium account).
My set up is a Raspberry Pi 4 8GB connected to HDMI running 64 bit OS 
HOST OS VERSION
balenaOS 2.87.16+rev1 (production)
It has booted correctly with the Balena logo showing on screen. I can see it listed as online in my Balena dashbord. All services are running.
Target supervisor state
{
  "local": {
    "name": "long-fin",
    "config": {
      "RESIN_SUPERVISOR_DELTA_VERSION": "3",
      "RESIN_SUPERVISOR_NATIVE_LOGGER": "true",
      "RESIN_HOST_CONFIG_arm_64bit": "1",
      "RESIN_HOST_CONFIG_avoid_warnings": "1",
      "RESIN_HOST_CONFIG_disable_splash": "1",
      "RESIN_HOST_CONFIG_dtoverlay": "\"vc4-fkms-v3d\"",
      "RESIN_HOST_CONFIG_dtparam": "\"i2c_arm=on\",\"spi=on\",\"audio=on\"",
      "RESIN_HOST_CONFIG_gpu_mem": "16",
      "RESIN_HOST_FIREWALL_MODE": "",
      "RESIN_SUPERVISOR_DELTA": "1",
      "RESIN_SUPERVISOR_POLL_INTERVAL": "900000"
    },
    "apps": {
      "1876447": {
        "name": "balena-sound",
        "services": {},
        "networks": {},
        "volumes": {}
      }
    }
  },
  "dependent": {
    "apps": {},
    "devices": {}
  }
}
I am new to Linux  and raspberry Pi .

Getafix17, 2021-12-04:
I tried using another card with a fresh install and have the same issues.

npmstart-pray, 2021-12-05:
HifiBerryOS. Especially if you're using a DAC on your Pi.

Getafix17, 2021-12-06:
> HifiBerryOS. Especially if you're using a DAC on your Pi.

I am not using a DAC on the 8GB Pi I have for every day use. It is connected to a monitor via HDMI.  I have another 2GB Pi with Volumio and Moode audio that work without any problems. I was interested in testing multiroom playback with Balena sound. Both the Production and Development mode don't seem to work for me using Pi4 (64 bit OS).

npmstart-pray, 2021-12-06:
There’s currently a problem with services starting. The lead dev has assured me he’ll turn his attentions to it this month - it used to work (I set up 2 systems as late as July), but then something changed on their end. 
I believe the other OS I mentioned will work as you hope, with or without a DAC. 

Sent from nowhere near my ready room with my new tricorder

Getafix17, 2021-12-06:
Ok thanks for the heads-up.

tmigone, 2022-01-24:
Hey @Getafix17 based on the supervisor state you shared it looks like you did not push a balenaSound release to your devices. If you are new I suggest you check out [this guide](https://www.balena.io/docs/learn/getting-started/raspberrypi3/nodejs/) to get familiar with the terminology or [this other one](https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/) which covers how to deploy balenaSound.

If you are stuck or having issues with the workflow I'd recommend posting to https://forums.balena.io/, we've got several support agents during the day scouring the forums and helping out users, you'll get a much faster response over there. GitHub issues we typically use to track outstanding bugs/improvements.

closed 2022-01-24 by tmigone