iotsound/iotsound #526: bluetooth: unreliable bluetooth pairing status on balenaSound

tmigone, 2021-12-21:
**Describe the problem**

Devices that have been previously unpaired are still retained and listed as if they were paired. This can lead to devices not being able to re-pair. Manually removing the devices using `bluetoothctl` or `btmgmt` fixes the issue.

This probably needs to be fixed upstream: https://github.com/balenablocks/bluetooth

petrosagg, 2022-01-03:
I was facing the same problem on my bluetooth speaker setup where my bluetooth agent was: `bt-agent --capability NoInputNoOutput`

I fixed this problem by switching the parameters to `bt-agent --capability DisplayOnly --pin ./pins.conf` where pins.conf contains `* *` (accept any pin for any device). Now it doesn't matter how many times I unpair my phone, I can always re-pair it.

The technique is described here https://forums.raspberrypi.com/viewtopic.php?t=304434

I haven't tested it but it might be as simple as changing this line https://github.com/balenablocks/bluetooth/blob/master/bluetooth-agent#L28

rcooke-warwick, 2022-04-05:
I was able to reproduce, creating upstream issue here: https://github.com/balenablocks/bluetooth/issues/28