iotsound/iotsound #527: AptX, AptX HD & AAC support

tommyzat, 2021-12-22:
Hello!

I'm wondering if BalenaSound will provide support for AptX/OpenAptX and if not, can that be implemented in a custom manner?

tmigone, 2021-12-23:
Hi @tommyzat, support for this is coming when we upgrade the audio service to use PulseAudio v15. Earlier today I started the [PR](https://github.com/balenablocks/audio/tree/84-bump-pa-version) for that, testing is going to take a few weeks with holidays coming up but I definitely want to get it done during January.

tommyzat, 2021-12-23:
Thank you! Very exciting news

tommyzat, 2022-01-29:
Hello! Connecting to the raspberry pi does not prompt my phone that AptX is available. This should be happening.

npmstart-pray, 2022-02-01:
Is it available on your phone?
For me, it's a hard pass, and should be for you too: https://www.cnet.com/tech/mobile/aptx-everything-you-need-to-know-about-the-wireless-bluetooth-enhancement/
FLAC is (and has been for some time)the clear way to go.

tommyzat, 2022-02-02:
@npmstart-pray i didn't understand what you wrote; does AptX work for you? What does FLAC have to do with bluetooth streaming? That doesn't exist for bluetooth.

npmstart-pray, 2022-02-02:
You’re asking for further data compression for an interface when there is already data compression used by the streaming platform. You need to ask Spotify/Apple Music/Tidal etc to encode their content using AptX, and their platform to determine that your streamer is capable of decoding that algorithm - balena can’t do anything to help unless/until data comes from upstream in AptX format, but your hardware running balena also has to support that codec/data compression algorithm. 

Explain what you’re looking for from AptX integration? How do you want it used?

Sent from nowhere near my ready room with my new tricorder

tommyzat, 2022-02-02:
What does the streaming source have anything to do with bluetooth streaming from a device to balena? Regardless if I use either Spotify or my own local FLAC music there's going to be a bottleneck via the bluetooth connection, and I'd like to minimise it by using AAC/Apt X which should be superior to SBC.

npmstart-pray, 2022-02-02:
https://www.cnet.com/tech/mobile/aptx-everything-you-need-to-know-about-the-wireless-bluetooth-enhancement/

What you’re asking for is the balena devs to write code to detect AptX data streams coming from Bluetooth devices to your balena sound streamer, yes?  The source must be sending AptX encoded data. Does your intended source relay AptX data over its Bluetooth transmitter? If it does, then the client must recognize it, and I’m not sure a RasPi does, or balena or whatever DAC can decode it back to analog waveforms. 

Sent from nowhere near my ready room with my new tricorder

commit 7994635e (2022-01-10): audio: update to latest audio block running PA v15

Closes: #527, #300
Change-type: minor