iotsound/iotsound #542: RPi Zero W with Adafruit bonnet with no audio

ssombra, 2022-01-24:
Just installed balena sound on a RPi Zero W with an Adafruit speaker bonnet following the online instructions. Everything seems ok, the services are running, but I don't get any audio. I checked the troubleshooting page but no luck. I tried both the spotify cllient and the airplay client and none of them works: I can select the "BALENASOUND SPOTIFY" option in the Spotify app, but when hitting play there is not audio. In addition to that, I can see that the progess bar sometimes goes back one second and continues playing. 
any idea what am I doing wrong?

TurtleRad, 2022-01-24:
Same problem here. Tried it out on RPi3 and RPi4. I can output from my devices to BalenaSound on bluetooth, Airplay and SpotifyConnect; there's just no sound exiting the Pi's.

jannis3005, 2022-01-25:
Same here, just updated from 03f4b6b to 961b818, and audio stopped working

closed 2022-02-03 by bulldozer-balena[bot]

tmigone, 2022-02-03:
Should be fixed by https://github.com/balenalabs/balena-sound/pull/547, update to v3.8.4 to get the fix!

commit 0240cc6f (2022-02-03): audio: fix no sound bug

Fix bug introduced by upgrading PulseAudio to v15

Closes: #542, #543, #544, #545
Change-type: patch
PR #547 (2022-02-03): audio: fix no sound bug
Fix a bug introduced by upgrading audio block to latest version running PulseAudio v15.

PulseAudio v14 introduced significant changes to stream behavior when changing default sink (described here: https://www.freedesktop.org/wiki/Software/PulseAudio/Notes/14.0/#significantroutingchangestodefaultsinkssources). 

Closes: #542, #543, #544, #545
Change-type: patch