iotsound/iotsound #543: No Audio output on HiFiBerry DAC Plus on Latest Release ( 4575313 )

magauthority, 2022-01-25:
I upgraded my Pi to the latest release 4575313 and am getting no sound output at all through either spotify connect, or airplay. 
Downgrading the pi to a previous release resolved the issue and all works as expected. 

**Device information**
 - Board Raspberry Pi 3B+
 - Device type: Raspberry Pi 3
 - balenaOS 2.83.21+rev1

**Additional context**
Using HiFIBerry DAC Plus setting - has been working perfectly for a few years now.

fgebhart, 2022-01-28:
I'm facing the same issue with a Raspberry Pi 4B. Tried the following balenaOS versions (and Release), without success:
* raspberrypi4-64-2.85.2+rev3-v12.10.10 (414bed8)
* raspberrypi4-64-2.88.5+rev1-v12.11.0 (414bed8)
* raspberrypi4-64-2.75.0+rev1-v12.5.10 (414bed8)
* raspberrypi4-64-2.41.0+rev4-v10.2.2 (414bed8)

Also tried the headphone jack and a HifiBerry MiniAmp DAC.

@magauthority Any pointers on how to downgrade to previous releases? In the balena dashboad I can see `Track latest` or `Pin 414bed8` only.

dey-d, 2022-01-29:
we're all having the same problems over at balena forums: https://forums.balena.io/t/no-sound-after-fresh-install/351047/6

magauthority, 2022-01-29:
@fgebhart because I've been using Balena sound for a while, previous releases showed up in my dashboard, so I just had to select a previous one to downgrade. I did search on how to select a specific release, but came up empty on that. Maybe someone else can comment on how to do that until this issue with the latest release gets sorted out?

tmigone, 2022-02-03:
Should be fixed by https://github.com/balenalabs/balena-sound/pull/547, update to v3.8.4 to get the fix!

closed 2022-02-03 by tmigone