iotsound/iotsound #544: RPI3B or RPI -> No sound whatsoever

Mehrdadg, 2022-01-29:
Hi 
I am new to this BalenaSound but I have been trying for the past many hours to get BalenaSound working on multiple RPIs without any success. 
I have tried all the trick that was able to find googling (forcing audio to HDMI, 3.5mm jack, deleting vc4-fkms-v3d on Rpi3, ...) and none so far fixed my issue.

I use the cloud app to create an image (for both RPI3 and 4) and go through installing without any issues. I can use my ipad and see the newly configured BalenaSound in my Airplay menus and when I select it, it connects without any issues but there is no sound (yes I even tried increasing the sound on IPAD to max as someone had suggested).

Anyhow, it looks like a pretty good software but for the life of me I can't get it to work as I get no sound!

Any help is greatly appreciated.
thanks

achochia, 2022-01-29:
Can confirm same warnings on 2 fresh 32Bit BalenaSound installs on Raspberry Pi 3b+, and on a Raspberry Pi ZeroW with a supported DAC. Been having these issues with the newest version and even using the older 2.38.0 rev1 on my 3B+. Attempts with a 64Bit OS had the same issues, and no fixes specified online helped.

Mehrdadg, 2022-01-29:
Achochia- The error messages I posted earlier are from fresh install on RPI3B+. I don't have RPI ZeroW to try and don't have DAC.
I am just relying on the Headphone Jack from the RPIs and/or HDMI outputs.
one issue that was raised along ago was related to 64b OS on RPI3 which the recommendation was to remove "vc4-fkms-v3d" from the device which did nothing for me

achochia, 2022-01-29:
Im just adding more devices and variations to the mix. I had used BalenaSound for years now and this is the first time a fresh install gives me this much headache. I had a working version from a build in September, which correspond to tag ca96642. I think ill try to fork that revision and attempt from there.

TurtleRad, 2022-01-30:
I would suggest merging this with #543 and #542 … Seems the latest update is breaking the audio outputs on RPi.

Maximus48p, 2022-01-31:
Confirm no sound output 3.5mm jack on Pi3b+ with AUDIO_OUTPUT on RPI_HEADPHONES.
version 2.83.21+rev1 and 12.11.30.

Patrick114, 2022-02-02:
Having this issue as well, rolling back to 3.6.19 fixed it

Mehrdadg, 2022-02-03:
Stupid question... I looked everywhere to see how I can downgrade to version 3.6.19 but I don't see this option available to me.
Any hind is greatly appreciated.
thanks

tmigone, 2022-02-03:
Should be fixed by https://github.com/balenalabs/balena-sound/pull/547, update to v3.8.4 to get the fix!

closed 2022-02-03 by tmigone