iotsound/iotsound #546: Apt-X/AAC not working

tommyzat, 2022-02-03:
Hello!

BalenaSound uses pulseaudio v15, however there Apt-X is not functional.

npmstart-pray, 2022-02-03:
Have you tried the HD Version?

tommyzat, 2022-02-04:
I hope you're a really good troll (see here https://github.com/balenalabs/balena-sound/issues/527#issuecomment-1028290266), but please don't litter github issues unless you have information based on your past experience or you can actually help with the issue.

npmstart-pray, 2022-02-04:
Maybe it's a shairport issue?

tommyzat, 2022-02-05:
Please stop replying

npmstart-pray, 2022-02-05:
You can close this if you opened it. 

Sent from nowhere near my ready room with my new tricorder

the4anoni, 2022-02-28:
I can confirm, that AptX and AAC doesn't work. Only SBC works.

tommyzat, 2022-08-21:
This is solved by installing gst-plugins-bad in the audio block

reg98643t, 2023-06-10:
> This is solved by installing gst-plugins-bad in the audio block

Hey man, how do you install plugins to the audio block?

tommyzat, 2023-06-10:
> > This is solved by installing gst-plugins-bad in the audio block
> 
> Hey man, how do you install plugins to the audio block?

Check out the audio block from balena's github page, build it and use it in this project with the changes I mentioned