iotsound/iotsound #551: Use Analog Audio from e.g. Hifi as input

ColynDee, 2022-02-09:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
An option to use analog audio input into the Raspberry PI via USB soundcard or similar in addition to Bluetooth, Airplay etc.

**Describe alternatives you've considered**
Can't think of an alternative practical solution, but open to ideas.

**Additional context**

samaero, 2022-02-17:
This is possible using the variable SOUND_ENABLE_SOUNDCARD_INPUT = 1

ColynDee, 2022-02-19:
Thank you. I am very new to all this. So your response is greatly appreciated. I have the USB soundcard working now for output, but cannot get the audio in on the soundcard to work. The logs show that it appears to be active, but .....

Any ideas? 
Many thanks

samaero, 2022-02-19:
I tested on my system (running BalenaSound 3.8.5) and I'm actually no longer able to get my input working, so I wonder if it was broken in a recent update. (I did also switch hardware to a Pi Zero 2 W, so that could also be causing the issue.)

The BalenaSound team always called soundcard input an experimental feature, so it's not officially supported, but has worked perfectly in previous versions, so hopefully it'll be possible to get working again.

ColynDee, 2022-02-20:
I have the same version 93.8.50 as you. I've tried with both a PI 3b and a PI4 - same result. I don't think it is experimental anymore. looking at the docs: "USB Soundcards
USB soundcards are also supported without any special configuration needed. Just make sure you power cycle your device after plugging in the soundcard and you should be good to go."

Is there a forum where we can ask about this, or is this it?

samaero, 2022-02-20:
Using USB sound cards as outputs is supported, but using it as input is experimental. There is a forum that you can try posting to [here](https://forums.balena.io/c/project-help/balenasound), but I don't think the devs have the bandwidth to respond at the moment, and there's not a lot of activity from advanced users who actually know what they're doing!

ColynDee, 2022-02-21:
I had a look on the forum, and found something that was very useful. It looks like the usb soundcard was not being selected. I have the input working now. If you run alsamixer from a terminal in the audio block, and then select the usb soundcard, unmute the mic, set levels etc., all works fine. The only problem is that this is not permanent. So you need to set this every time, which isn't ideal; but, at least it is working. Thanks for all your help. really appreciate it.

knutsoned, 2023-08-11:
I'm using a HiFiBerry DAC+ ADC with a RasPi 4B. I added the hifiberry-dacplusadc dtoverlay and disabled onboard audio. ALSA is detecting the device but the name of the source it creates is alsa_input.dac.stereo-fallback instead of alsa_input.dac.analog-stereo. I set SOUND_ENABLE_SOUNDCARD_INPUT = 1 as a device variable. After editing start.sh L45 to read:

local INPUT_DEVICE_FULLNAME="alsa_input.$INPUT_DEVICE.stereo-fallback"

I am still not able to get the sound connected to the line input jack to work

dotzeno, 2023-11-22:
I'd like to do this with a RASPIAUDIO ULTRA+. anyone have experience or advice with this?