iotsound/iotsound #556: Avoid using network_mode: host for containers

tommyzat, 2022-02-21:
Is it possible to avoid this for all containers? I can't communicate between any other containers when using network_mode: host

tmigone, 2022-02-21:
Hey @tommyzat, requirement for host mode boils down to plugins needing it to function properly. It should be possible theoretically to remove the host mode requirement by providing net/admin adequate privileges, but it's honestly a hassle to setup/test and the only time I tried with shairport container it stopped working when they updated the binary.

That said, you still should be able to communicate between containers no matter what network mode they are in. Can you tell me more about your set up? What type of comms are you attempting? host to bridge? bridge to host?

tommyzat, 2022-02-21:
Thanks a lot! I managed to get it working by accesing localhost instead of docker name and adding to ports:
    ports:
      - "127.0.0.1:1234:1234"

closed 2022-02-21 by tommyzat