iotsound/iotsound #558: Why does using belena-sound require a cloud account?

joxl, 2022-02-27:
**Is your feature request related to a problem? Please describe.**
For a completely LAN-isolated audio solution, I see no reason for any portion of it to ever need internet connectivity (with the exception of internet audio sources, which is solely user preference and does not require a balenaSound cloud account).

**Describe the solution you'd like**
A balena-sound installation literally has one or more Raspberry Pis running on a LAN. The local devices should be able to host all services needed for deployment and setup.

**Describe alternatives you've considered**
Perhaps there are important "features" of balena-sound that require a cloud-based "arbiter" of sorts (e.g. push notifications, access to the system while not on the LAN, etc). If that is the case then a suitable alternative would be to offer a deployment option that opts out of those features with the benefit of a purely LAN-only solution (i.e. cloud account not required).

**Additional context**
The very first line of the **Motivation** section in the balena-sound [README.md](https://github.com/balenalabs/balena-sound/blob/65ee51a622365abd6ae2463fe56d3bb583cd4635/README.md#motivation) states:

> There are many commercial solutions out there that provide functionality similar to balenaSound. Most of them though come with a premium price tag and are riddled with privacy concerns.

Yet the first prerequisite to installation is a cloud account with a 3rd party vendor. This is a huge red flag to anyone with genuine privacy concerns.

eiddor, 2022-02-27:
AFAIK, you only need a balena account in order to use balena tools to deploy the software to a device.  There should be nothing stopping you from manually setting this up with Docker using the files in the repo.

npmstart-pray, 2022-02-27:
Or just use HifiBerryOS. If you don't need the networking aspect, being a standalone unit is probably the best for you.

joxl, 2022-02-28:
> There should be nothing stopping you from manually setting this up with Docker using the files in the repo.

Thanks @eiddor, I came here and glanced at the features and thought everything looked great, but the "advertising" didn't make it sound like that would be possible. This is really helpful.

joxl, 2022-02-28:
> If you don't need the networking aspect, being a standalone unit is probably the best for you.

Thanks for this, @npmstart-pray. I've also looked at that and I'm leaning the same way. Much appreciated.

tmigone, 2022-04-01:
Hi @joxl thanks for raising this.

We agree that you shouldn't need a balenaCloud account to operate a device running this project. For that reason last year we introduced the concept of [openFleets on balenaHub](https://www.balena.io/blog/introducing-open-fleets-and-self-submitted-apps-and-blocks-on-balenahub). This lets you deploy balenaSound to a device without needing a balenaCloud account, link for the balenaSound open fleet is here: https://hub.balena.io/balenalabs/balenasound. That said, balenaHub and openFleets are products we are still working on and there are several known issues with them, for example when your device boots it takes several minutes to download and bootstrap the application and end user doesn't get any feedback during that time which is bad. We are working on improving a lot of things however we don't have a clear timeframe of when most of this will be completed so that is why you might have not seen a lot of fuss about this.

Going back to the privacy concerns, openFleets still rely on our infrastructure and platform mostly for fleet management (for example to get software updates) so the devices are still connected to our API for these purposes. As you can see the project's code is open source however part of our API code is not. I understand you might have concerns over that so there are two options here:
- modify the project to run on a non-balena device using docker. In fact that's something we once had instructions for, but over the course of the years we've added functionality that relies on balenaOS features so we removed them because we couldn't maintain them.
- use the unmanaged variant of balenaOS (getting started guide here: https://www.balena.io/os/docs/raspberrypi3/getting-started/). Unmanaged balenaOS is a version of balenaOS that's not connected to our API, it's essentially a standalone device. 

At the moment these are the options we have available. We'd like to offer docs that cover them in detail however it's not something we've had the time to prioritize so far.