iotsound/iotsound #559: Run BalenaSound in a docker on Kubuntu and 3 clients on Rasp B+

MrSaatci, 2022-03-03:
Hi, 
I'm trying to get this set up running but I keep hitting a wall. 
I'm trying to have multi-room set up with balena sound and I had 3 old pi's, when I try to flash them with BalenaOS and BalenaSound they seemed to be unable to handle both client and master processes, so I thought I could have BalenaSound running on a dedicated Linux laptop that I use to run HomeAssistant (Hassio), which runs on a VM. 

I downloaded all the folders for Balena Sound and pulled the image for ResinOS, but when I try to build it I get an error building audio saying that my dockerfile is missing although it is there

```
docker-compose up -d --build

Creating network "balena-sound-master2_default" with the default driver
Creating volume "balena-sound-master2_spotifycache" with default driver
Building audio
ERROR: Cannot locate specified Dockerfile: Dockerfile```

Any ideas? I'm a noob in docker, so the answer might be obvious but I've been looking for about 3 hours online and nothing seems to work

tmigone, 2022-04-05:
@MrSaatci, the problem you are having is most likely related to docker not knowing how to handle `Dockerfile.template` files. This is an abstraction that we built for balena engine and not something docker engine knows how to process.

At the moment balenaSound is not really meant to be run on hosts that are not running balenaOS. Initially we did support installing it on generic hosts using docker but since then the project has evolved and complex features we've added made it drift from that possibility. 

Regarding your specific error, you can try renaming the `Dockerfile.template` files into `Dockerfile` and replace any instance of `%%BALENA_ARCH%%` or `%%BALENA_MACHINE_NAME` variables with the appropriate value, `amd64` for example is a good start. That said, you would be treading on uncharted waters, it's likely that several other modifications are needed in order for the project to run on a generic linux machine and you might even need to remove entire features. 

I can try and help you out if you decide to give it a shot, but it's not something we are looking to prioritize ourselves.

ThePoonerCap, 2023-11-03:
Its this still the case? I have multiple docker servers that i would love to use instead of buying more pis just because of this. 
Thanks !