iotsound/iotsound #563: Raspberry Pi Zero W and Adafruit I2S DAC Bonnet

tromano32, 2022-03-17:
Hi, I am having trouble getting audio out of the dac on my Pi Zero W with the Adafruit I2S Audio Bonnet - UDA1334A

I deployed from
https://hub.balena.io/balenalabs/balenasound

I entered the below configuration per the https://sound.balenalabs.io/docs/audio-interfaces/#dac-boards page. 
"hifiberry-dac”,“i2s-mmap”

and still no sound.

Please advise

Thanks !

tmigone, 2022-04-05:
hi @tromano32 were you able to solve this? If not can you try posting to https://forums.balena.io/? That's a more adequate channel for this type of troubleshooting, you'll get far better response times over there.

closed 2022-04-05 by tmigone

tromano32, 2022-04-05:
I have given up. It appears the rasp pi zero w is just too slow … the dac bonnet works but playback stutters so bad it’s not usable even with extras plug ins like Bluetooth and multi room disabled 

Thanks anyway 

Sent from my mobile.