iotsound/iotsound #576: no sound on jack

Moule0169, 2022-05-04:
Hello,
I am a new user of Balena Sound, which I have just installed on a Raspberry Pi 4B.
OS 2.95.8
Supervisor 12.1.38

The services are active (except the hostname), I can connect via spotify, but I have sound only on the HDMI0 port, not on the 3.5mm jack, nor on the HDMI1.
I was in version 3.8.5, I tried to upgrade to 3.8.4 because I saw that a bug had been solved this way in February.
I also forced the AUDIO_OUTPUT variable to RPI_HEADPHONES, but it doesn't change anything, the sound is still coming out of the HDMI0 plug.

Can anyone help me?
Thank you,

MahmudThomas, 2022-05-18:
I am also having this issue. I've been looking through the issues tab and the belena forums and I cant find a response for this question. If I find one I will link it here.

Blazin64, 2022-05-22:
@Moule0169 @hcwf @MahmudThomas @tommyzat

I was having a similar issue, but have audio working after making some changes. I selected my Raspberry Pi device from the balena cloud dashboard and made the following changes:

---

**Device variables:**

| Name         | Value          |
|--------------|----------------|
| AUDIO_OUTPUT | RPI_HEADPHONES |
| SOUND_VOLUME | 100            |

**Device configuration:**

Delete the `vc4-kms-v3d` entry from "Define DT overlays".

(Found in  [this thread](https://forums.balena.io/t/raspberry-pi-4-no-3-5mm-sound-using-balenasound/356827).)

---

After a reboot, I had working Spotify and Bluetooth audio. I hope this also works for your devices! 😀

hcwf, 2022-05-26:
@Blazin64 Thank you, deleting that entry actually worked! My sound is working now.

pechano, 2022-11-17:
Thank you Blazin64! Your solution also worked for me.

Bananz0, 2024-06-22:
Doing this on my Pi4 makes it spam some errors and using this "vc4-kms-v3d-pi4" fixes it.