iotsound/iotsound #579: Multiroom-client error - Failed to connect, connection refused, error in socket

jjotto, 2022-06-12:
Device Information:
Board:  Intel NUC NUC5i5MYBE
Type:  Intel NUC
BalenaOS:  2.98.33
Supervisor:  14.0.3

I'm getting a series of errors that show this information in the logs.  I've tried to re-install BalenaOS and Supervisor a few times and tried a few different versions.  

multiroom-client [Error] (Connection) Failed to connect to host 'localhost', error: Connection refused
multiroom-client [Error] (Controller) Error: Connection refused
multiroom-client [Error] (Connection) Error in socket shutdown: Socket not connected

[image]

[image]

frindaddy, 2022-12-09:
I am also consistently getting this error, although on a Raspberry Pi 4. I believe this results in my multiroom-client cutting in and out during Spotify play.

jkr442, 2023-02-27:
I am getting the same issue on my Intel NUC. I don't have the option of ethernet where this device is to be used.

IIxauII, 2023-03-03:
Running into the same issue, pi zero w setup as MULTI_ROOM_CLIENT - wifi connection

MaximUltimatum, 2024-04-09:
Googling around and delving through the forums. I think this issue should have been fixed in #466
See [forum post here](https://forums.balena.io/t/my-belenas-forget-that-they-know-each-other/323239)
Other thing to check would be make sure you don't have client device isolation enabled on the wifi network. This is easy to check in Unifi routers on the wifi settings page.

All that said, I just upgraded my network from 3's to 4's and now I'm having this issue. Will try deleting existing fleet and recreating a new one to see if that helps