iotsound/iotsound #583: SOUND_VOLUME ignored on pi 3B+

jbudding, 2022-06-16:
Hello,

I have a pi 3B+ and have set SOUND_VOLUME to 85 and the volume is still too low out the headphone jack.  If I Terminal into the audio container 
and run alsamixer the GUI always shows 75 as the volume level.  I raise it to 85 and it works.  Reboot and it reverts to 75.
release is: 471209f2ec1ec90d5ba9df3b5a031540
and list the environment variables in the audio container I see:
...truncated
SOUND_VOLUME=85
RESIN_DEVICE_ARCH=armv7hf
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket
SOUND_MODE=STANDALONE
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
BALENA_DEVICE_TYPE=raspberrypi3
RESIN_SERVICE_KILL_ME_PATH=/tmp/balena/handover-complete
RESIN_HOST_OS_VERSION=balenaOS 2.98.33
RESIN_DEVICE_NAME_AT_INIT=vast-engine
RESIN_DEVICE_TYPE=raspberrypi3
BALENA_HOST_OS_VERSION=balenaOS 2.98.33
BALENA_SERVICE_HANDOVER_COMPLETE_PATH=/tmp/balena/handover-complete
... 
even though alsamixer reverts to 75 after a reboot.

johan489, 2023-09-19:
I have the same issue on my pi4+.
[balenaOS 2.105.1+rev1]
URRENT RELEASE
[3.9.3]

Matts-V, 2023-11-03:
Having the same issue here pi4+