iotsound/iotsound #589: Spotify doesn't work anymore

fuzzybear62, 2022-07-28:
**Describe the problem**
Spotify stop working, no sound at all and song skipping on the spotify app

**Device information**
 - Balena-fin 1.1

grubersh, 2022-07-28:
Yep. Same here and same error with balenaOS 2.99.27.  Error from logs:

` spotify  [2022-07-28T18:19:39Z INFO  librespot_playback::player] Loading <Temple of Love - 1992> with Spotify URI <spotify:track:53O11dqezFoFYaaMjr1SPP>
 spotify  [2022-07-28T18:19:39Z ERROR librespot_core::channel] channel error: 2 1
 spotify  [2022-07-28T18:19:39Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
 spotify  [2022-07-28T18:19:39Z WARN  librespot_playback::player] Unable to load <SpotifyId { id: 221052802452224273845189659844783082157, audio_type: Track }>
 spotify      Skipping to next track`

grubersh, 2022-07-28:
seems to be tracked here: https://github.com/librespot-org/librespot/issues/972

fuzzybear62, 2022-07-29:
On another client I use https://github.com/spocon/spocon and it works fine

andrewn, 2022-08-01:
This is fixed in `librespot@0.4.2` so as a temporary workaround [I forked this repo](https://github.com/andrewn/balena-sound/tree/bug/spotify-connect-issue) and pushed an upgraded librespot docker image.

AmauryBJ, 2022-08-01:
@andrewn thanks ! How can i deploy the update ?

andrewn, 2022-08-02:
> @andrewn thanks ! How can i deploy the update ?

If you check out the [forked repository](https://github.com/andrewn/balena-sound/tree/bug/spotify-connect-issue), then run [`balena push`](https://www.balena.io/docs/learn/deploy/deployment/#balena-push) to upload the forked code to your fleet.

This is just temporary while the main repository catches up as I can't offer any support.

jonaskor, 2022-08-07:
Blocking `apresolve.spotify.com` in your router config can be a temporary fix, too.

maggie0002, 2022-08-11:
@andrewn, thanks for the fix. 

I'm thinking it may be easier to integrate if we were to move the contents of your forked repo in to this one. So instead of relying on `tmigone/librespot:...` which draws on an external image, we move that build process in to this repo which will make it easier to maintain and keep up to date. Would you be able to look at a pull request for that?

andrewn, 2022-11-10:
@maggie0002 I'd be happy to give this a go.

I'm not sure the best way to structure this, so can you point me to some examples please?

maggie0002, 2022-11-10:
@andrewn, that's great news, I would have had no way to test this. Really appreciate it. 

It seems like your fix is to use a forked version of the package and bump the version inside: https://github.com/andrewn/balena-sound/commit/c6eba2c96d81063d9b02c652e03b9e43a82e50c? 

So the current Dockerfile in this repo (https://github.com/balena-labs-projects/balena-sound/blob/master/plugins/spotify/Dockerfile.template) looks like this:

```
FROM tmigone/librespot:0.3.1-pulseaudio

ENV PULSE_SERVER=tcp:localhost:4317

COPY start.sh /usr/src/

CMD [ "/bin/bash", "/usr/src/start.sh" ]
```

I don't think we need to maintain `librespot:0.3.1-pulseaudio` as a separate package anymore, we can bring it in to this repo and maintain it here. So the Dockerfile (https://github.com/balena-labs-projects/balena-sound/blob/master/plugins/spotify/Dockerfile.template) would change to something like this, and hopefully will all work ok?

```
ARG BALENA_ARCH=%%BALENA_ARCH%%

# Build process from: https://git.alpinelinux.org/aports/tree/testing/librespot/APKBUILD
FROM balenalib/$BALENA_ARCH-alpine:edge as librespot-builder
WORKDIR /app

ARG LIBRESPOT_VERSION=0.4.2

RUN install_packages alsa-lib-dev pulseaudio-dev cargo curl

RUN curl -sL "https://github.com/librespot-org/librespot/archive/refs/tags/v${LIBRESPOT_VERSION}.tar.gz" --output librespot.tar.gz && \
    mkdir /app/librespot-src && \
    tar -zxvf librespot.tar.gz --directory /app/librespot-src --strip-components=1

WORKDIR /app/librespot-src

ENV CARGO_HOME=/app/cargo
ENV RUSTFLAGS="-C target-feature=-crt-static"

RUN cargo build \
    --release \
    --features alsa-backend,pulseaudio-backend \
    --verbose

FROM balenalib/$BALENA_ARCH-alpine:3.14-run
WORKDIR /app

ENV PULSE_SERVER=tcp:localhost:4317

RUN install_packages libgcc alsa-lib-dev pulseaudio-dev

COPY --from=librespot-builder /app/librespot-src/target/release/librespot /usr/bin/librespot

COPY start.sh /usr/src/

CMD [ "/bin/bash", "/usr/src/start.sh" ]

```

PR #610 (2022-11-30): Build librespot 0.4.2 in container (fixes #589)
Spotify playback is broken in librespot 0.3.1 and depending on tmigone/librespot:0.3.1-pulseaudio means we can't easily upgrade to the latest version.

Building within the project allows us to choose which version is tracked by balena-sound, [as suggested here](https://github.com/balena-labs-projects/balena-sound/issues/589#issuecomment-1310072186).

I pushed this to a balena device and it builds correctly and plays Spotify.

Fixes #589
commit 7c8bb5b2 (2022-11-30): Build librespot 0.4.2 in container (fixes #589)

Spotify playback is broken in librespot 0.3.1 and depending on tmigone/librespot:0.3.1-pulseaudio means we can't easily upgrade to the latest version. Building within the project allows us to choose which version is tracked by balena-sound
closed 2022-11-30 by maggie0002