iotsound/iotsound #594: No sound on RPi 4 3.5mm connector

bennetrr, 2022-09-19:
**Describe the problem**
After deploying I don't get any sound through the 3.5mm connector on the Raspberry Pi.
I did not change any settings (so multiroom and automatic output are enabled)
[balenasound.log](https://github.com/balenalabs/balena-sound/files/9600451/balenasound.log)

What is weird is that the audio container only detected two HDMI audio cards, but the 3.5mm connector works (I tried it with another software).

**Device information**
 - Raspberry Pi 4B 8GB
 - 16 GB microSDHC Class 10
 - `balenaOS v2.103.1+rev1` (Production Mode)
- Supervisor `v14.0.14`

ciaranowens, 2022-09-29:
Looks related to this: 

https://github.com/balenablocks/audio/issues/100

bennetrr, 2022-10-08:
Thanks!
As the issue suggested, just moving the lines 

```
# Enable VC4 Graphics
dtoverlay=vc4-kms-v3d
```

to the end of the file fixes the problem.

closed 2022-10-08 by bennetrr